cloud-native-robotz-hackathon/infrastructure #4: Automation Cloud OCP Setup

goetzrieger, 2024-01-23:
Use DevSecOps WS Automation to:
- Install Gitea
- Install ODF & Quay
- Skupper
- DevSpaces

rbo, 2024-01-23:
Skupper connection is missing.

rbo, 2024-06-04:
[image]

Installed operators:
```
[lab-user@bastion ~]$ oc get installplan -A
NAMESPACE             NAME            CSV                                            APPROVAL    APPROVED
gitea-operator        install-zslmv   gitea-operator.v2.0.1                          Automatic   true
openshift-devspaces   install-524m8   devworkspace-operator.v0.27.0-0.1714987663.p   Automatic   true
openshift-operators   install-26vsv   openshift-gitops-operator.v1.12.3              Automatic   true
openshift-operators   install-jckp8   web-terminal.v1.10.0                           Automatic   true
openshift-operators   install-qmgtf   openshift-pipelines-operator-rh.v1.14.4        Automatic   true
openshift-storage     install-mk55v   ocs-operator.v4.15.2-rhodf                     Automatic   true
redhat-ods-operator   install-vj2wq   rhods-operator.2.8.2                           Automatic   true
```

rbo, 2024-06-19:
Skupper is handled via #15 and #16

closed 2024-06-19 by rbo