cloud-native-robotz-hackathon/infrastructure #22: Preapre  OpenShit AI env per user

rbo, 2024-06-20:
* Deploy pipeline & co per user https://github.com/mamurak/os-mlops

nexus-Six, 2024-07-04:
---
kind: Secret
apiVersion: v1
metadata:
  name: aws-connection-object-detection
  namespace: object-detection
  labels:
    opendatahub.io/dashboard: 'true'
    opendatahub.io/managed: 'true'
  annotations:
    opendatahub.io/connection-type: s3
    openshift.io/display-name: object-detection
stringData:
  AWS_ACCESS_KEY_ID: minio
  AWS_DEFAULT_REGION: us-east-1
  AWS_S3_BUCKET: object-detection
  AWS_S3_ENDPOINT: http://minio-models-service.minio.svc:9000
  AWS_SECRET_ACCESS_KEY: minio123
type: Opaque

---
kind: PersistentVolumeClaim
apiVersion: v1
metadata:
  name: object-detection-training-pvc
  namespace: object-detection
spec:
  accessModes:
    - ReadWriteOnce
  resources:
    requests:
      storage: 10Gi
  volumeMode: Filesystem

rbo, 2024-07-04:
> ```yaml
> ---
> kind: Secret
> apiVersion: v1
> metadata:
>   name: aws-connection-object-detection
>   namespace: object-detection
>   labels:
>     opendatahub.io/dashboard: 'true'
>     opendatahub.io/managed: 'true'
>   annotations:
>     opendatahub.io/connection-type: s3
>     openshift.io/display-name: object-detection
> stringData:
>   AWS_ACCESS_KEY_ID: minio
>   AWS_DEFAULT_REGION: us-east-1
>   AWS_S3_BUCKET: object-detection
>   AWS_S3_ENDPOINT: http://minio-models-service.minio.svc:9000
>   AWS_SECRET_ACCESS_KEY: minio123
> type: Opaque
> ```

This is already available as `workbench-bucket`

https://github.com/cloud-native-robotz-hackathon/agnosticd/blob/development/ansible/roles_ocp_workloads/ocp4_workload_cloud_native_robot/tasks/workload-per-team.yml#L15-L83
https://github.com/cloud-native-robotz-hackathon/agnosticd/blob/development/ansible/roles_ocp_workloads/ocp4_workload_cloud_native_robot/defaults/main.yaml#L18