cloud-native-robotz-hackathon/infrastructure #31: OpenShift AI 2.5+ does not work because of  Error 1064: You have an error in your SQL synta during pipeline impor

rbo, 2024-07-03:
Log from pipeline server `oc logs ds-pipeline-dspa-*`

```
I0703 12:43:36.730078       1 error.go:278] Error 1064: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near 'OVER (PARTITION BY PipelineId ORDER BY CreatedAtInSec DESC) rn FROM pipeline_...' at line 1
InternalServerError: Failed to execute SQL for listing pipelines
github.com/kubeflow/pipelines/backend/src/common/util.NewInternalServerError
	/opt/app-root/src/backend/src/common/util/error.go:144
github.com/kubeflow/pipelines/backend/src/apiserver/storage.(*PipelineStore).ListPipelinesV1
	/opt/app-root/src/backend/src/apiserver/storage/pipeline_store.go:244
github.com/kubeflow/pipelines/backend/src/apiserver/resource.(*ResourceManager).ListPipelinesV1
	/opt/app-root/src/backend/src/apiserver/resource/resource_manager.go:260
github.com/kubeflow/pipelines/backend/src/apiserver/server.(*PipelineServer).listPipelines
	/opt/app-root/src/backend/src/apiserver/server/pipeline_server.go:381
github.com/kubeflow/pipelines/backend/src/apiserver/server.(*PipelineServer).ListPipelinesV1
	/opt/app-root/src/backend/src/apiserver/server/pipeline_server.go:430
github.com/kubeflow/pipelines/backend/api/v1beta1/go_client._PipelineService_ListPipelinesV1_Handler.func1
	/opt/app-root/src/backend/api/v1beta1/go_client/pipeline.pb.go:2108
main.apiServerInterceptor
	/opt/app-root/src/backend/src/apiserver/interceptor.go:30
github.com/kubeflow/pipelines/backend/api/v1beta1/go_client._PipelineService_ListPipelinesV1_Handler
	/opt/app-root/src/backend/api/v1beta1/go_client/pipeline.pb.go:2110
google.golang.org/grpc.(*Server).processUnaryRPC
	/remote-source/deps/gomod/pkg/mod/google.golang.org/grpc@v1.56.3/server.go:1335
google.golang.org/grpc.(*Server).handleStream
	/remote-source/deps/gomod/pkg/mod/google.golang.org/grpc@v1.56.3/server.go:1712
google.golang.org/grpc.(*Server).serveStreams.func1.1
	/remote-source/deps/gomod/pkg/mod/google.golang.org/grpc@v1.56.3/server.go:947
runtime.goexit
	/usr/lib/golang/src/runtime/asm_amd64.s:1598
ResourceManager (v1beta1): Failed to list pipelines with context &{0xc00109ae20}, options &{20 0xc00069ee00}
github.com/kubeflow/pipelines/backend/src/common/util.(*UserError).wrapf
	/opt/app-root/src/backend/src/common/util/error.go:266
github.com/kubeflow/pipelines/backend/src/common/util.Wrapf
	/opt/app-root/src/backend/src/common/util/error.go:337
github.com/kubeflow/pipelines/backend/src/apiserver/resource.(*ResourceManager).ListPipelinesV1
	/opt/app-root/src/backend/src/apiserver/resource/resource_manager.go:262
github.com/kubeflow/pipelines/backend/src/apiserver/server.(*PipelineServer).listPipelines
	/opt/app-root/src/backend/src/apiserver/server/pipeline_server.go:381
github.com/kubeflow/pipelines/backend/src/apiserver/server.(*PipelineServer).ListPipelinesV1
	/opt/app-root/src/backend/src/apiserver/server/pipeline_server.go:430
github.com/kubeflow/pipelines/backend/api/v1beta1/go_client._PipelineService_ListPipelinesV1_Handler.func1
	/opt/app-root/src/backend/api/v1beta1/go_client/pipeline.pb.go:2108
main.apiServerInterceptor
	/opt/app-root/src/backend/src/apiserver/interceptor.go:30
github.com/kubeflow/pipelines/backend/api/v1beta1/go_client._PipelineService_ListPipelinesV1_Handler
	/opt/app-root/src/backend/api/v1beta1/go_client/pipeline.pb.go:2110
google.golang.org/grpc.(*Server).processUnaryRPC
	/remote-source/deps/gomod/pkg/mod/google.golang.org/grpc@v1.56.3/server.go:1335
google.golang.org/grpc.(*Server).handleStream
	/remote-source/deps/gomod/pkg/mod/google.golang.org/grpc@v1.56.3/server.go:1712
google.golang.org/grpc.(*Server).serveStreams.func1.1
	/remote-source/deps/gomod/pkg/mod/google.golang.org/grpc@v1.56.3/server.go:947
runtime.goexit
	/usr/lib/golang/src/runtime/asm_amd64.s:1598
Failed to list pipelines (v1beta1) in namespace . Check error stack
github.com/kubeflow/pipelines/backend/src/common/util.(*UserError).wrapf
	/opt/app-root/src/backend/src/common/util/error.go:266
github.com/kubeflow/pipelines/backend/src/common/util.Wrapf
	/opt/app-root/src/backend/src/common/util/error.go:337
github.com/kubeflow/pipelines/backend/src/apiserver/server.(*PipelineServer).ListPipelinesV1
	/opt/app-root/src/backend/src/apiserver/server/pipeline_server.go:432
github.com/kubeflow/pipelines/backend/api/v1beta1/go_client._PipelineService_ListPipelinesV1_Handler.func1
	/opt/app-root/src/backend/api/v1beta1/go_client/pipeline.pb.go:2108
main.apiServerInterceptor
	/opt/app-root/src/backend/src/apiserver/interceptor.go:30
github.com/kubeflow/pipelines/backend/api/v1beta1/go_client._PipelineService_ListPipelinesV1_Handler
	/opt/app-root/src/backend/api/v1beta1/go_client/pipeline.pb.go:2110
google.golang.org/grpc.(*Server).processUnaryRPC
	/remote-source/deps/gomod/pkg/mod/google.golang.org/grpc@v1.56.3/server.go:1335
google.golang.org/grpc.(*Server).handleStream
[... 46 lines cut ...]
    return self.request("GET", url,
  File "/opt/app-root/lib64/python3.9/site-packages/kfp_server_api/rest.py", line 224, in request
    raise ApiException(http_resp=r)
kfp_server_api.exceptions.ApiException: (500)
Reason: Internal Server Error
HTTP response headers: HTTPHeaderDict({'content-length': '1131', 'content-type': 'application/json', 'date': 'Wed, 03 Jul 2024 12:13:33 GMT', 'gap-auth': 'system:serviceaccount:team-1-ai:model@cluster.local', 'gap-upstream-address': 'localhost:8888', 'set-cookie': '8b9bbb891eb973b7ce5013a9117aa392=3e9ac1b4b5fe46034b6deb89950e4404; path=/; HttpOnly; Secure; SameSite=None'})
HTTP response body: {"error":"Failed to list pipelines (v1beta1) in namespace . Check error stack: ResourceManager (v1beta1): Failed to list pipelines with context \u0026{0xc00103a520}, options \u0026{20 0xc00056e600}: InternalServerError: Failed to execute SQL for listing pipelines: Error 1064: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near 'OVER (PARTITION BY PipelineId ORDER BY CreatedAtInSec DESC) rn FROM pipeline_...' at line 1","code":13,"message":"Failed to list pipelines (v1beta1) in namespace . Check error stack: ResourceManager (v1beta1): Failed to list pipelines with context \u0026{0xc00103a520}, options \u0026{20 0xc00056e600}: InternalServerError: Failed to execute SQL for listing pipelines: Error 1064: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near 'OVER (PARTITION BY PipelineId ORDER BY CreatedAtInSec DESC) rn FROM pipeline_...' at line 1","details":[{"@type":"type.googleapis.com/google.rpc.Status","code":13,"message":"Internal Server Error"}]}

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/app-root/lib64/python3.9/site-packages/tornado/web.py", line 1786, in _execute
    result = await result
  File "/opt/app-root/lib64/python3.9/site-packages/elyra/pipeline/handlers.py", line 160, in post
    response = await PipelineProcessorManager.instance().process(pipeline)
  File "/opt/app-root/lib64/python3.9/site-packages/elyra/pipeline/processor.py", line 105, in process
    res = await asyncio.get_event_loop().run_in_executor(None, processor.process, pipeline)
  File "/usr/lib64/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/opt/app-root/lib64/python3.9/site-packages/elyra/pipeline/kfp/processor_kfp.py", line 297, in process
    raise RuntimeError(
RuntimeError: Failed to get ID of Kubeflow pipeline: 'model-training-cpu' - Check Kubeflow Pipelines runtime configuration: 'odh_dsp
```

[image]

rbo, 2024-07-03:
Work-a-round: downgrade to OpenShift AI 2.5

rbo, 2024-07-04:
With OpenShift AI 2.8 to 2.9+ there was a breaking change in kubeflow pipeline "area". We have to use a new Workbench base image as well: 
* https://github.com/mamurak/os-mlops/blob/main/container-images/object-detection/Containerfile-workbench
*  quay.io/mmurakam/workbenches:object-detection-v2.0.0	
  * v2 is for >=2.9
  * v1 is for <=2.8

mamurak, 2024-08-16:
Fixing this in #61.

rbo, 2024-08-19:
Let's try to update to OpenShift AI 2.11....  Only dev env.

rbo, 2024-08-19:
Branch `dev-rhoai-211` created in repos:
 * https://github.com/cloud-native-robotz-hackathon/object-detection-notebooks
 * https://github.com/cloud-native-robotz-hackathon/infrastructure
 * https://github.com/cloud-native-robotz-hackathon/lab-guide
Integration tests.

rbo, 2024-08-19:
* Fixed object-detection imagestream (was not select able)
 * Looks like pipeline server is not working:

During pipeline start:
```
Traceback (most recent call last):
  File "/opt/app-root/lib64/python3.9/site-packages/elyra/pipeline/kfp/processor_kfp.py", line 219, in process
    client = ArgoClient(
  File "/opt/app-root/lib64/python3.9/site-packages/kfp/_client.py", line 196, in __init__
    if not self._context_setting['namespace'] and self.get_kfp_healthz(
  File "/opt/app-root/lib64/python3.9/site-packages/kfp/_client.py", line 406, in get_kfp_healthz
    raise TimeoutError(
TimeoutError: Failed getting healthz endpoint after 5 attempts.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/app-root/lib64/python3.9/site-packages/tornado/web.py", line 1790, in _execute
    result = await result
  File "/opt/app-root/lib64/python3.9/site-packages/elyra/pipeline/handlers.py", line 160, in post
    response = await PipelineProcessorManager.instance().process(pipeline)
  File "/opt/app-root/lib64/python3.9/site-packages/elyra/pipeline/processor.py", line 105, in process
    res = await asyncio.get_event_loop().run_in_executor(None, processor.process, pipeline)
  File "/usr/lib64/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/opt/app-root/lib64/python3.9/site-packages/elyra/pipeline/kfp/processor_kfp.py", line 238, in process
    raise RuntimeError(
RuntimeError: Failed to initialize `kfp.Client()` against: 'https://ds-pipeline-pipelines-definition-team-1-ai.apps.cluster-bch2f.bch2f.sandbox1300.opentlc.com' - Check Kubeflow Pipelines runtime configuration: 'odh_dsp' - [TIP: did you mean to set 'https://ds-pipeline-pipelines-definition-team-1-ai.apps.cluster-bch2f.bch2f.sandbox1300.opentlc.com/pipeline' as the endpoint, take care not to include 's' at end]
```
[image]

At the Pipelines overview: `Pipeline version cannot be rendered`

rbo, 2024-08-19:
Looks like it can not connect to https://ds-pipeline-pipelines-definition-team-2-ai.apps.cluster-bch2f.bch2f.sandbox1300.opentlc.com/ because of oauth.
[fedora-detection-0-fedora-detection (1).log](https://github.com/user-attachments/files/16664980/fedora-detection-0-fedora-detection.1.log)

Fix of pipeline server deployment is on the way.

rbo, 2024-08-21:
Looks all good. Let's bring all together and stay with RHOAI 2.11

 * https://github.com/cloud-native-robotz-hackathon/object-detection-notebooks => https://github.com/cloud-native-robotz-hackathon/object-detection-notebooks/pull/3
 * https://github.com/cloud-native-robotz-hackathon/infrastructure => https://github.com/cloud-native-robotz-hackathon/infrastructure/pull/63
 * https://github.com/cloud-native-robotz-hackathon/lab-guide => https://github.com/cloud-native-robotz-hackathon/lab-guide/pull/3

rbo, 2024-08-22:
Fixed with upgrade to 2.11

closed 2024-08-22 by rbo