cloud-native-robotz-hackathon/infrastructure #38: Try to use a build and use small Triton image

rbo, 2024-07-04:
At then moment, Triton image 10g big. 

@goetzrieger please provide information to the small available version of triton

rbo, 2024-07-31:
Right now, we are using:
 * nvcr.io/nvidia/tritonserver:24.06-py3 7.64 GB

The minimal is a bit less
 * nvcr.io/nvidia/tritonserver:24.06-py3-min 3.85 GB

Let's try to build an own only with onnx backend:

https://docs.nvidia.com/deeplearning/triton-inference-server/user-guide/docs/customization_guide/compose.html

rbo, 2024-07-31:
PS.: We need a smaller image. I tried to deploy the triton server via GitOps on a new/empty robot. It failed:
```
Events:
  Type     Reason   Age                   From     Message
  ----     ------   ----                  ----     -------
  Warning  Failed   34m                   kubelet  Failed to pull image "nvcr.io/nvidia/tritonserver:24.06-py3": rpc error: code = Unknown desc = Error writing blob: error storing blob to file "/var/tmp/storage235593114/1": error happened during read: unexpected EOF
  Warning  Failed   8m20s (x8 over 3h9m)  kubelet  Error: ErrImagePull
  Warning  Failed   8m20s                 kubelet  Failed to pull image "nvcr.io/nvidia/tritonserver:24.06-py3": rpc error: code = Unknown desc = Error writing blob: error storing blob to file "/var/tmp/storage322025009/3": error happened during read: unexpected EOF
  Normal   Pulling  8m9s (x9 over 3h32m)  kubelet  Pulling image "nvcr.io/nvidia/tritonserver:24.06-py3"
  ```

goetzrieger, 2024-08-14:
Build fails whatever I try, opened a Triton Server issue

https://github.com/triton-inference-server/server/issues/7513

rbo, 2024-09-17:
done via pre-pull #56

closed 2024-09-17 by rbo