cloud-native-robotz-hackathon/infrastructure #40: Build a new robot base image with

rbo, 2024-07-04:


rbo, 2024-07-26:
Image: https://drive.google.com/file/d/1u6H2ehd_MgcmpHC4bgbLNpBMzQoW4SK2/view?usp=drive_link

Changelog & how to write to SD Card: https://docs.google.com/document/d/1o5QDPy3MtI5fAXJiuUbeHjMPbMgLx7oMNrzNkmNlNIQ/edit?usp=drive_link

## ToDo:

- [ ] Add a version file into the image. For Example `/robot-image-version` content `robot-hackathon-image.20240724` (CC: @goetzrieger )

goetzrieger, 2024-07-26:
20240726: https://drive.google.com/file/d/139K2DgZnrxKIiAU-ErjdPXQHGoGOXubV/view?usp=drive_link

rbo, 2024-07-26:
LGTM

```
root@data:~# hostname -f
data.lan
root@data:~# oc get nodes
NAME   STATUS   ROLES    AGE   VERSION
data   Ready    <none>   21m   v1.21.0
root@data:~# cat /
.disk/               dev/                 lib/                 mnt/                 robot-image-version  sbin/                sys/                 var/
bin/                 etc/                 lost+found/          opt/                 root/                snap/                tmp/
boot/                home/                media/               proc/                run/                 srv/                 usr/
root@data:~# cat /robot-image-version
robot-hackathon-image.20240726
root@data:~# podman ps
CONTAINER ID  IMAGE       COMMAND     CREATED     STATUS      PORTS       NAMES
root@data:~# podman images
REPOSITORY                               TAG                            IMAGE ID      CREATED      SIZE
quay.io/microshift/flannel-cni           4.8.0-0.okd-2021-10-10-030117  0e66d6f50c69  2 years ago  8.78 MB
quay.io/microshift/service-ca-operator   4.8.0-0.okd-2021-10-10-030117  0d3ab4435626  2 years ago  276 MB
quay.io/microshift/kube-rbac-proxy       4.8.0-0.okd-2021-10-10-030117  7f149e453e90  2 years ago  41.5 MB
quay.io/microshift/hostpath-provisioner  4.8.0-0.okd-2021-10-10-030117  fdef3dc1264a  2 years ago  39.3 MB
quay.io/microshift/haproxy-router        4.8.0-0.okd-2021-10-10-030117  37292c44812e  2 years ago  225 MB
quay.io/microshift/coredns               4.8.0-0.okd-2021-10-10-030117  67a95c8f1590  2 years ago  265 MB
quay.io/microshift/cli                   4.8.0-0.okd-2021-10-10-030117  33a276ba2a97  2 years ago  205 MB
quay.io/microshift/flannel               4.8.0-0.okd-2021-10-10-030117  85fc911ceba5  3 years ago  68.1 MB
k8s.gcr.io/pause                         3.5                            f7ff3c404263  3 years ago  491 kB
root@data:~# oc get pods
No resources found in default namespace.
root@data:~# oc get pods -A
NAMESPACE                       NAME                                  READY   STATUS    RESTARTS   AGE
kube-system                     kube-flannel-ds-xzb5p                 1/1     Running   0          22m
kubevirt-hostpath-provisioner   kubevirt-hostpath-provisioner-7dmzv   1/1     Running   0          22m
openshift-dns                   dns-default-pddnk                     2/2     Running   0          22m
openshift-dns                   node-resolver-29k2l                   1/1     Running   0          22m
openshift-ingress               router-default-85bcfdd948-lh6lv       1/1     Running   0          22m
openshift-service-ca            service-ca-7764c85869-6mfbx           1/1     Running   0          22m
root@data:~#
```

From my point of view we can close the Issue.

rbo, 2024-07-26:
I don't know if we need the triton pre-pull stuff. Because from workflow the robots will run anyway before the workshops starts and get a default deployment of triton via [robot-gitops](https://github.com/cloud-native-robotz-hackathon/robot-gitops) repo. So it will pulled anyway quite early

goetzrieger, 2024-07-30:
But I would hate if we run in any network performance issues when 10 robots download 11GB at the same time... or the nvidia registry is having issues. If we would mirror it locally somehow...

rbo, 2024-07-31:
From my perspective, we can close this issue.

closed 2024-07-31 by rbo