cloud-native-robotz-hackathon/infrastructure #56: Can not pull  nvcr.io/nvidia/tritonserver:24.06-py3 on robots anymore

rbo, 2024-07-31:
```
root@data:~# podman pull nvcr.io/nvidia/tritonserver:24.06-py3
Trying to pull nvcr.io/nvidia/tritonserver:24.06-py3...
Getting image source signatures
Copying blob d5a2ad729c09 skipped: already exists
Copying blob 04674895a676 skipped: already exists
Copying blob 8031832c8ce4 skipped: already exists
Copying blob b4104c2b7496 skipped: already exists
Copying blob 4f4fb700ef54 skipped: already exists
Copying blob 341e2429afc2 done
Copying blob afc8d30ee6fd [=============>------------------------] 1.1GiB / 3.0GiB
Copying blob beee5165c2db done
Copying blob b8b59560bcf4 done
Copying blob da77e32b0054 done
Copying blob 2c38b9cb8c50 done
Copying blob 43220f522638 done
Copying blob 305c28fb3f16 done
Copying blob d70e86ed1828 done
Copying blob 7c61d096c130 done
Copying blob f6b9bfc97b66 done
Copying blob cecedd88a50d [==================================>---] 526.0MiB / 571.8MiB
Copying blob 14334dd017c7 done
Copying blob 686d947cff1a done
Copying blob 6398cd9b4ef7 done
Copying blob 4f4fb700ef54 skipped: already exists
Copying blob fcef1900e41a done
Copying blob 4f4fb700ef54 skipped: already exists
Copying blob fd5090e5783a [========>-----------------------------] 572.0MiB / 2.4GiB
Copying blob 4f4fb700ef54 skipped: already exists
Copying blob e98f5ef4ac1f done
Copying blob f3978d285fd1 done
Error: writing blob: storing blob to file "/var/tmp/storage727061959/2": happened during read: unexpected EOF
```

Gitops triton deployment failed as well. 

Two possible options:
1) Try a smaller image #38 
2) pre-pull the image looks like is not an option...

nexus-Six, 2024-08-28:
Just copied to Triton image over to our Quay org repo. Pulling the image is now faster and more stable.

@rbo Please change the pull references where required :

quay.io/cloud-native-robotz-hackathon/tritonserver:24.08-py3

rbo, 2024-08-28:
Triton image updated https://github.com/cloud-native-robotz-hackathon/robot-gitops/commit/0bc0ddf62f9fddefc3a11d0d672d9982bcae1a50

rbo, 2024-08-28:
```
root@data:~# time podman pull quay.io/cloud-native-robotz-hackathon/tritonserver:24.08-py3
Trying to pull quay.io/cloud-native-robotz-hackathon/tritonserver:24.08-py3...
Getting image source signatures
Copying blob 4f4fb700ef54 skipped: already exists
Copying blob 4f4fb700ef54 skipped: already exists
Copying blob 4f4fb700ef54 skipped: already exists
Copying blob 345022223d5f done
Copying blob 4673bd629234 done
Copying blob b32b4ed9fe13 done
Copying blob 49f79dddbde4 done
Copying blob 441074d4d2bc done
Copying blob e63ce922f022 done
Copying blob 859c4bf73ee0 done
Copying blob 468f39f5e8c4 done
Copying blob 75b00786e09f done
Copying blob e478db066bbf done
Copying blob 632454d400d3 done
Copying blob d3b35ba05b07 done
Copying blob b21bf2a9afa9 done
Copying blob e0b7299c849b done
Copying blob 77a288066c2f done
Copying blob 036a7953bb7d done
Copying blob 4a87c76a537c done
Copying blob f8cff84b5e43 done
Copying blob 4f4fb700ef54 skipped: already exists
Copying blob 05c1fdf2f90d done
Copying blob ad7aee037f00 done
Copying blob 4f4fb700ef54 skipped: already exists
Copying blob 6a836d4c5e08 done
Copying blob 4f4fb700ef54 skipped: already exists
Copying blob 4c7aae254fff done
Copying blob e2f3a5ae8147 done
Copying config d874e371e7 done
Writing manifest to image destination
Storing signatures
d874e371e7177e8d04d88d942c8e852ceb234d2e1699bda9273f027e3cb2dd19

real	33m42.212s
user	23m10.805s
sys	11m25.959s
root@data:~#
```

Good news pull works, it was not measureable but i was around 50/50 between pull and untar filesystem layers.

nexus-Six, 2024-09-05:
Tested on the robot with Argo and Microshift. Takes about 30 min but works.
Some prepulling on the robots would still be good.

rbo, 2024-09-17:
I tried with podman save & load it took to much time.

Solution:
 * Connect a RJ45 wire from robot to router/network
 * Login via ssh & wire connection
 * Disable wifi interface `ip link set dev wlan0 down`
 * Pull the image `podman pull quay.io/cloud-native-robotz-hackathon/tritonserver:24.08-py3`
 * Remove wire and reboot

rbo, 2024-09-17:
Works pretty well, we can close it.

rbo, 2024-09-18:
Pulled image at gort is gone. Looks like the pre-pull is not very persistent

closed 2024-09-19 by rbo