cloud-native-robotz-hackathon/infrastructure #58: robot-app pipeline failed

rbo, 2024-08-12:
```
The node was low on resource: ephemeral-storage. Threshold quantity: 16015370671, available: 12622432Ki. Container step-build-and-push was using 8447016Ki, request is 0, has larger consumption of ephemeral-storage.
```

rbo, 2024-08-13:
New cluster, and try to build https://github.com/cloud-native-robotz-hackathon/starter-app-python/commit/f67feb32ae33908bc18d0849cb8bcd24122c7911 failed on x64 nodes as well.

FYI, size of the arm image:
```
default-route-openshift-image-registry.apps.cluster-bdw98.bdw98.sandbox1004.opentlc.com/team-1-dev/starter-app-python  test-arm64  092eac33ae58  28 minutes ago  1.57 GB
```

rbo, 2024-08-13:
Added 
```
      computeResources:
        requests:
          ephemeral-storage: "1Gi"
        limits:
          ephemeral-storage: "10Gi"
```

Pod Evicted as well.

```
The node was low on resource: ephemeral-storage. Threshold quantity: 16015370671, available: 14083620Ki.

```

How to fetch ephemeral-storage from node:
```
kubectl get --raw "/api/v1/nodes/<node>/proxy/stats/summary" | jq '.pods[] | select(.podRef.name=="pod-name") '
```

rbo, 2024-08-13:
I'm pretty sure the build was running like a charm when I write the pipeline. It must be a dependency from the requirements.txt - shipped a new bigger version?

For example the arm images from @goetzrieger : https://quay.io/repository/grieger/starter-app-python-archtest?tab=tags is 305MB and my one around 1.57 GB!

rbo, 2024-08-13:
Here the log of the build: 
```
$ oc logs -f release-tag-hjgv4m-build-on-amd64-pod
Defaulted container "step-build-and-push" out of: step-build-and-push, prepare (init), place-scripts (init), working-dir-initializer (init)
STEP 1/9: FROM ubi8/python-38:1-129
Resolving "ubi8/python-38" using unqualified-search registries (/etc/containers/registries.conf)
Trying to pull registry.access.redhat.com/ubi8/python-38:1-129...
Getting image source signatures
Checking if image destination supports signatures
Copying blob sha256:6f08d340e5b4c55a33e7b01004d2461a05fb099090542ba4b235071591d994b1
Copying blob sha256:d96c2f4a5465b79c577fdecc27cd49a2c0f103fd2cfd372e667512be069e0143
Copying blob sha256:6c53be4efe39270422e3e2f7ee1c15887955e3d5e378ea6577c622f358f87691
Copying blob sha256:1a3c9e41511310aa895b5bc5b140efe6b6d9e9583f3ce9d70205d1f415d2d25f
Copying config sha256:7eccbd7f5c3017c77b3223cb885f66835ddbb8bdb252523af535768c164a03d7
Writing manifest to image destination
Storing signatures
STEP 2/9: ENV PORT 5000
STEP 3/9: EXPOSE 5000
STEP 4/9: WORKDIR /usr/src/app
STEP 5/9: COPY requirements.txt .
STEP 6/9: RUN pip install --no-cache-dir -r requirements.txt
Collecting gunicorn
  Downloading gunicorn-23.0.0-py3-none-any.whl (85 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 85.0/85.0 kB 3.8 MB/s eta 0:00:00
Collecting Flask
  Downloading flask-3.0.3-py3-none-any.whl (101 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 101.7/101.7 kB 12.3 MB/s eta 0:00:00
Collecting requests
  Downloading requests-2.32.3-py3-none-any.whl (64 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 64.9/64.9 kB 215.5 MB/s eta 0:00:00
Collecting opencv-python-headless
  Downloading opencv_python_headless-4.10.0.84-cp37-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (49.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 49.9/49.9 MB 211.5 MB/s eta 0:00:00
Collecting torch
  Downloading torch-2.4.0-cp38-cp38-manylinux1_x86_64.whl (797.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 797.2/797.2 MB 75.4 MB/s eta 0:00:00
Collecting torchvision
  Downloading torchvision-0.19.0-cp38-cp38-manylinux1_x86_64.whl (7.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 7.0/7.0 MB 78.6 MB/s eta 0:00:00
Collecting packaging
  Downloading packaging-24.1-py3-none-any.whl (53 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 54.0/54.0 kB 237.7 MB/s eta 0:00:00
Collecting Werkzeug>=3.0.0
  Downloading werkzeug-3.0.3-py3-none-any.whl (227 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 227.3/227.3 kB 297.9 MB/s eta 0:00:00
Collecting click>=8.1.3
  Downloading click-8.1.7-py3-none-any.whl (97 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 97.9/97.9 kB 298.8 MB/s eta 0:00:00
Collecting Jinja2>=3.1.2
  Downloading jinja2-3.1.4-py3-none-any.whl (133 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 133.3/133.3 kB 320.5 MB/s eta 0:00:00
Collecting blinker>=1.6.2
  Downloading blinker-1.8.2-py3-none-any.whl (9.5 kB)
Collecting itsdangerous>=2.1.2
  Downloading itsdangerous-2.2.0-py3-none-any.whl (16 kB)
Collecting importlib-metadata>=3.6.0
  Downloading importlib_metadata-8.2.0-py3-none-any.whl (25 kB)
Collecting certifi>=2017.4.17
  Downloading certifi-2024.7.4-py3-none-any.whl (162 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 163.0/163.0 kB 306.3 MB/s eta 0:00:00
Collecting idna<4,>=2.5
  Downloading idna-3.7-py3-none-any.whl (66 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 66.8/66.8 kB 279.3 MB/s eta 0:00:00
Collecting urllib3<3,>=1.21.1
  Downloading urllib3-2.2.2-py3-none-any.whl (121 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 121.4/121.4 kB 316.9 MB/s eta 0:00:00
Collecting charset-normalizer<4,>=2
  Downloading charset_normalizer-3.3.2-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (141 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 141.1/141.1 kB 327.1 MB/s eta 0:00:00
Collecting numpy>=1.17.3
[... 51 lines cut ...]
Collecting nvidia-nvjitlink-cu12
  Downloading nvidia_nvjitlink_cu12-12.6.20-py3-none-manylinux2014_x86_64.whl (19.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 19.7/19.7 MB 57.2 MB/s eta 0:00:00
Collecting pillow!=8.3.*,>=5.3.0
  Downloading pillow-10.4.0-cp38-cp38-manylinux_2_28_x86_64.whl (4.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.5/4.5 MB 62.9 MB/s eta 0:00:00
Collecting zipp>=0.5
  Downloading zipp-3.20.0-py3-none-any.whl (9.4 kB)
Collecting MarkupSafe>=2.0
  Downloading MarkupSafe-2.1.5-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (26 kB)
Collecting mpmath<1.4,>=1.1.0
  Downloading mpmath-1.3.0-py3-none-any.whl (536 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 536.2/536.2 kB 94.8 MB/s eta 0:00:00
Installing collected packages: mpmath, zipp, urllib3, typing-extensions, sympy, pillow, packaging, nvidia-nvtx-cu12, nvidia-nvjitlink-cu12, nvidia-nccl-cu12, nvidia-curand-cu12, nvidia-cufft-cu12, nvidia-cuda-runtime-cu12, nvidia-cuda-nvrtc-cu12, nvidia-cuda-cupti-cu12, nvidia-cublas-cu12, numpy, networkx, MarkupSafe, itsdangerous, idna, fsspec, filelock, click, charset-normalizer, certifi, blinker, Werkzeug, triton, requests, opencv-python-headless, nvidia-cusparse-cu12, nvidia-cudnn-cu12, Jinja2, importlib-metadata, gunicorn, nvidia-cusolver-cu12, Flask, torch, torchvision
Successfully installed Flask-3.0.3 Jinja2-3.1.4 MarkupSafe-2.1.5 Werkzeug-3.0.3 blinker-1.8.2 certifi-2024.7.4 charset-normalizer-3.3.2 click-8.1.7 filelock-3.15.4 fsspec-2024.6.1 gunicorn-23.0.0 idna-3.7 importlib-metadata-8.2.0 itsdangerous-2.2.0 mpmath-1.3.0 networkx-3.1 numpy-1.24.4 nvidia-cublas-cu12-12.1.3.1 nv [... cut ...] n-3.0.0 typing-extensions-4.12.2 urllib3-2.2.2 zipp-3.20.0

[notice] A new release of pip available: 22.2.2 -> 24.2
[notice] To update, run: pip install --upgrade pip
STEP 7/9: COPY . .
STEP 8/9: ENTRYPOINT ["python"]
STEP 9/9: CMD ["app.py"]
COMMIT image-registry.openshift-image-registry.svc.cluster.local:5000/team-1-dev/starter-app-python:23-amd64
time="2024-08-13T20:27:06Z" level=error msg="signal: terminated"
time="2024-08-13T20:27:06Z" level=error msg="(Unable to determine exit status)"
```

rbo, 2024-08-14:
Let's check version:

```
~ % podman run -ti --rm --entrypoint bash quay.io/grieger/starter-app-python-archtest:latest
(app-root) bash-4.4$ pip list
Package            Version
------------------ ---------
blinker            1.6.2
certifi            2023.7.22
charset-normalizer 3.2.0
click              8.1.6
Flask              2.3.2
gunicorn           21.2.0
idna               3.4
importlib-metadata 6.8.0
itsdangerous       2.1.2
Jinja2             3.1.2
MarkupSafe         2.1.3
packaging          23.1
pip                22.2.2
requests           2.31.0
setuptools         41.6.0
urllib3            2.0.4
Werkzeug           2.3.6
zipp               3.16.2

[notice] A new release of pip available: 22.2.2 -> 24.2
[notice] To update, run: pip install --upgrade pip
(app-root) bash-4.4$
```

Local build of commit `488867a61e8bdd4e40fe976f61e1416f6e59a479`
``` 
starter-app-python %  podman run -ti --rm --entrypoint bash  localhost/starter-app-python
(app-root) bash-4.4$ pip list
Package                Version
---------------------- ---------
blinker                1.8.2
certifi                2024.7.4
charset-normalizer     3.3.2
click                  8.1.7
filelock               3.15.4
Flask                  3.0.3
fsspec                 2024.6.1
gunicorn               23.0.0
idna                   3.7
importlib_metadata     8.2.0
itsdangerous           2.2.0
Jinja2                 3.1.4
MarkupSafe             2.1.5
mpmath                 1.3.0
networkx               3.1
numpy                  1.24.4
opencv-python-headless 4.10.0.84
packaging              24.1
pillow                 10.4.0
pip                    22.2.2
requests               2.32.3
setuptools             41.6.0
sympy                  1.13.2
torch                  2.4.0
torchvision            0.19.0
typing_extensions      4.12.2
urllib3                2.2.2
Werkzeug               3.0.3
zipp                   3.20.0

[notice] A new release of pip available: 22.2.2 -> 24.2
[notice] To update, run: pip install --upgrade pip
(app-root) bash-4.4$

```

Images size
```
starter-app-python % podman images
REPOSITORY                                   TAG         IMAGE ID      CREATED        SIZE
localhost/starter-app-python                 latest      44c39cdcb2a4  2 minutes ago  1.57 GB
quay.io/grieger/starter-app-python-archtest  latest      906fe50623eb  12 months ago  919 MB
registry.access.redhat.com/ubi8/python-38    1-129       9a48e9949d98  13 months ago  911 MB
starter-app-python %

```

rbo, 2024-08-14:
> quay.io/grieger/starter-app-python-archtest:latest
Is missing some depps. looks like that's tooo old. 

> quay.io/cloud-native-robotz-hackathon/starter-app-python:master
Is to old as well.

Begin of Jul I created the build pipeline, during that timeframe the build works quite well: https://github.com/cloud-native-robotz-hackathon/starter-app-python/commit/aaff68aef0134c7bf723279a78d2b605760686b3

Meanwhile, only torch and torch vision shipped a new release.
Let's pin the older version:
```
torch==2.3.1
torchvision==0.18.1
```

Local test build don't install nvidia cuda stuff anymore:

```
Installing collected packages: mpmath, zipp, urllib3, typing-extensions, sympy, pillow, packaging, numpy, networkx, MarkupSafe, itsdangerous, idna, fsspec, filelock, click, charset-normalizer, certifi, blinker, Werkzeug, requests, opencv-python-headless, Jinja2, importlib-metadata, gunicorn, torch, Flask, torchvision
Successfully installed Flask-2.3.2 Jinja2-3.1.4 MarkupSafe-2.1.5 Werkzeug-3.0.3 blinker-1.8.2 certifi-2024.7.4 charset-normalizer-3.3.2 click-8.1.7 filelock-3.15.4 fsspec-2024.6.1 gunicorn-21.2.0 idna-3.7 importlib-metadata-8.2.0 itsdangerous-2.2.0 mpmath-1.3.0 networkx-3.1 numpy-1.24.4 opencv-python-headless-4.10.0.84 packaging-24.1 pillow-10.4.0 requests-2.31.0 sympy-1.13.2 torch-2.3.1 torchvision-0.18.1 typing-extensions-4.12.2 urllib3-2.2.2 zipp-3.20.0
```

Let's test it via pipelines.

rbo, 2024-08-14:
Pipeline works again with the pinned versions. But the ephemeral-storage usage is massive:

```
while true; do kubectl get --raw "/api/v1/nodes/ip-10-0-30-216.us-east-2.compute.internal/proxy/stats/summary" | jq '.pods[] | select(.podRef.name=="release-tag-qdnr5-build-on-amd64-pod") | ."ephemeral-storage"' ; sleep 1; done
...

{
  "time": "2024-08-14T18:10:54Z",
  "availableBytes": 32411488256,
  "capacityBytes": 106769133568,
  "usedBytes": 8366813184,       => 8.3 GB!!!
  "inodesFree": 51176959,
  "inodes": 52166080,
  "inodesUsed": 134983
}

```

The image size is at the end around 1.5GB!

@nexus-Six in the call today, you mentioned a chance to get rid of torch? 

Additionally I changed the root disk size from 100g to 250 to have more headroom.

nexus-Six, 2024-08-15:
The torch calls need to be replaced by another package. Max is looking into this.

https://github.com/cloud-native-robotz-hackathon/starter-app-python/blob/master/lib/object_detection.py

nexus-Six, 2024-08-16:
Tested app start in DevSpaces and my custom pipeline -> both work now with the package reduced version
@rbo Please do a pipeline test in your environment

I still need to verify if the detection still works with the code changes, after I have a local setup running.

rbo, 2024-08-16:
Build on arm failed:
```
ERROR: Ignored the following versions that require a different python version: 1.25.0 Requires-Python >=3.9; 1.25.0rc1 Requires-Python >=3.9; 1.25.1 Requires-Python >=3.9; 1.25.2 Requires-Python >=3.9; 1.26.0 Requires-Python <3.13,>=3.9; 1.26.0b1 Requires-Python <3.13,>=3.9; 1.26.0rc1 Requires-Python <3.13,>=3.9; 1.26.1 Requires-Python <3.13,>=3.9; 1.26.2 Requires-Python >=3.9; 1.26.3 Requires-Python >=3.9; 1.26.4 Requires-Python >=3.9; 2.0.0 Requires-Python >=3.9; 2.0.0b1 Requires-Python >=3.9; 2.0.0rc1 Requires-Python >=3.9; 2.0.0rc2 Requires-Python >=3.9; 2.0.1 Requires-Python >=3.9; 2.1.0rc1 Requires-Python >=3.10
ERROR: Could not find a version that satisfies the requirement numpy==2.0.1 (from versions: 1.3.0, 1.4.1, 1.5.0, 1.5.1, 1.6.0, 1.6.1, 1.6.2, 1.7.0, 1.7.1, 1.7.2, 1.8.0, 1.8.1, 1.8.2, 1.9.0, 1.9.1, 1.9.2, 1.9.3, 1.10.0.post2, 1.10.1, 1.10.2, 1.10.4, 1.11.0, 1.11.1, 1.11.2, 1.11.3, 1.12.0, 1.12.1, 1.13.0, 1.13.1, 1.13.3, 1.14.0, 1.14.1, 1.14.2, 1.14.3, 1.14.4, 1.14.5, 1.14.6, 1.15.0, 1.15.1, 1.15.2, 1.15.3, 1.15.4, 1.16.0, 1.16.1, 1.16.2, 1.16.3, 1.16.4, 1.16.5, 1.16.6, 1.17.0, 1.17.1, 1.17.2, 1.17.3, 1.17.4, 1.17.5, 1.18.0, 1.18.1, 1.18.2, 1.18.3, 1.18.4, 1.18.5, 1.19.0, 1.19.1, 1.19.2, 1.19.3, 1.19.4, 1.19.5, 1.20.0, 1.20.1, 1.20.2, 1.20.3, 1.21.0, 1.21.1, 1.21.2, 1.21.3, 1.21.4, 1.21.5, 1.21.6, 1.22.0, 1.22.1, 1.22.2, 1.22.3, 1.22.4, 1.23.0rc1, 1.23.0rc2, 1.23.0rc3, 1.23.0, 1.23.1, 1.23.2, 1.23.3, 1.23.4, 1.23.5, 1.24.0rc1, 1.24.0rc2, 1.24.0, 1.24.1, 1.24.2, 1.24.3, 1.24.4)
ERROR: No matching distribution found for numpy==2.0.1
```

rbo, 2024-08-16:
```
% podman run -ti --rm registry.access.redhat.com/ubi8/python-38:1-129 pip install numpy==2.0.1
ERROR: Ignored the following versions that require a different python version: 1.25.0 Requires-Python >=3.9; 1.25.0rc1 Requires-Python >=3.9; 1.25.1 Requires-Python >=3.9; 1.25.2 Requires-Python >=3.9; 1.26.0 Requires-Python <3.13,>=3.9; 1.26.0b1 Requires-Python <3.13,>=3.9; 1.26.0rc1 Requires-Python <3.13,>=3.9; 1.26.1 Requires-Python <3.13,>=3.9; 1.26.2 Requires-Python >=3.9; 1.26.3 Requires-Python >=3.9; 1.26.4 Requires-Python >=3.9; 2.0.0 Requires-Python >=3.9; 2.0.0b1 Requires-Python >=3.9; 2.0.0rc1 Requires-Python >=3.9; 2.0.0rc2 Requires-Python >=3.9; 2.0.1 Requires-Python >=3.9; 2.1.0rc1 Requires-Python >=3.10
ERROR: Could not find a version that satisfies the requirement numpy==2.0.1 (from versions: 1.3.0, 1.4.1, 1.5.0, 1.5.1, 1.6.0, 1.6.1, 1.6.2, 1.7.0, 1.7.1, 1.7.2, 1.8.0, 1.8.1, 1.8.2, 1.9.0, 1.9.1, 1.9.2, 1.9.3, 1.10.0.post2, 1.10.1, 1.10.2, 1.10.4, 1.11.0, 1.11.1, 1.11.2, 1.11.3, 1.12.0, 1.12.1, 1.13.0, 1.13.1, 1.13.3, 1.14.0, 1.14.1, 1.14.2, 1.14.3, 1.14.4, 1.14.5, 1.14.6, 1.15.0, 1.15.1, 1.15.2, 1.15.3, 1.15.4, 1.16.0, 1.16.1, 1.16.2, 1.16.3, 1.16.4, 1.16.5, 1.16.6, 1.17.0, 1.17.1, 1.17.2, 1.17.3, 1.17.4, 1.17.5, 1.18.0, 1.18.1, 1.18.2, 1.18.3, 1.18.4, 1.18.5, 1.19.0, 1.19.1, 1.19.2, 1.19.3, 1.19.4, 1.19.5, 1.20.0, 1.20.1, 1.20.2, 1.20.3, 1.21.0, 1.21.1, 1.21.2, 1.21.3, 1.21.4, 1.21.5, 1.21.6, 1.22.0, 1.22.1, 1.22.2, 1.22.3, 1.22.4, 1.23.0rc1, 1.23.0rc2, 1.23.0rc3, 1.23.0, 1.23.1, 1.23.2, 1.23.3, 1.23.4, 1.23.5, 1.24.0rc1, 1.24.0rc2, 1.24.0, 1.24.1, 1.24.2, 1.24.3, 1.24.4)
ERROR: No matching distribution found for numpy==2.0.1

[notice] A new release of pip available: 22.2.2 -> 24.2
[notice] To update, run: pip install --upgrade pip
% **podman run -ti --rm registry.access.redhat.com/ubi9/python-312:1-20.1722518948 pip install numpy==2.0.1
Trying to pull registry.access.redhat.com/ubi9/python-312:1-20.1722518948...
Getting image source signatures
Checking if image destination supports signatures
Copying blob sha256:c6f5847a7d5ebb6d4fe76423af992ea0acefca3cbf204f5e1004116397e1f65e
Copying blob sha256:a81277bf03b3242dd8a344d908eaf743eeeec3855545305197e97f128248ca82
Copying blob sha256:679da2159f24d706632e5ced43625e0acf98deaea69b0a3d52848943764c39a1
Copying blob sha256:7594a545ac0ef3eb3c8367a508f6976ea1490cb83c32b11c01a87fef6723c8f7
Copying config sha256:a0b0aef9cdc623adc85a309495f9f5ae1379e285cfbff65230fa0eca5af94455
Writing manifest to image destination
Storing signatures
Collecting numpy==2.0.1
  Obtaining dependency information for numpy==2.0.1 from https://files.pythonhosted.org/packages/5e/e3/944b70438d3b7e2742fece7da8dfba6f7ef7dccdd163d1a613f7027f4d5b/numpy-2.0.1-cp312-cp312-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata
  Downloading numpy-2.0.1-cp312-cp312-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (62 kB)**
```

We should update the base image...

rbo, 2024-08-16:
Looks good for me.

Handover to @nexus-Six for final robot test.

nexus-Six, 2024-08-27:
Tested

closed 2024-08-27 by nexus-Six