cloud-native-robotz-hackathon/infrastructure #59: Robot camera produces images with red tint

goetzrieger, 2024-08-14:
At least ABC Warrior produces images with a red tint when using the test script.

rbo, 2024-08-14:
Terminator has the same issue
[image]

rbo, 2024-08-14:
Looks like the rasp. firmware is different

### Data - WORKS - new version

```
root@data:~# rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Wed Jan 11 17:40:52 UTC 2023 (1673458852)
    LATEST: Tue Jan 25 14:30:41 UTC 2022 (1643121041)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
root@data:~#
```
### Terminator - BROKEN - old version
```
root@terminator:~# rpi-eeprom-update
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
   CURRENT: Thu Sep  3 12:11:43 UTC 2020 (1599135103)
    LATEST: Tue Jan 25 14:30:41 UTC 2022 (1643121041)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
root@terminator:~#
```

```

rbo, 2024-08-14:
Different cameras:

[image]

@goetzrieger finding: 
> "NoIR board is a dark green coloured board, he normal is lighter brighter green colour"

rbo, 2024-09-16:
The following robots look good:

- camera-test-c3po.lan.jpg
- camera-test-data.lan.jpg
- camera-test-gort.lan.jpg
- camera-test-marc13.lan.jpg
- camera-test-marvin.lan.jpg
- camera-test-r2d2.lan.jpg
- camera-test-t-1000.lan.jpg

rbo, 2024-09-17:
I ordered new camera modules. (3x)