cloud-native-robotz-hackathon/infrastructure #66: Race Condition Issue on Interconnect Installation

nexus-Six, 2024-08-27:
When installing the Interconnect Operator via AgnosticD, the Controller Pod starts before the Route is present and the then fails permanently. Manual workaround ist to restart the Controller Pod.

This issue/workaround should be addressed by the cluster installation procedure.

rbo, 2024-08-29:
new test env, 
```
oc new-project red-hat-service-interconnect-data-center
oc apply -f - <<EOF
heredoc>           apiVersion: v1
          kind: ConfigMap
          metadata:
            name: skupper-site
            namespace: red-hat-service-interconnect-data-center
          data:
            cluster-permissions: "false"
            console: "true"
            console-authentication: internal
            console-password: ""
            console-user: ""
            enable-skupper-events: "true"
            flow-collector: "true"
            ingress: route
            name: red-hat-service-interconnect-data-center
            router-console: "false"
            router-logging: ""
            router-mode: interior
            service-controller: "true"
            service-sync: "true"
heredoc> EOF
configmap/skupper-site created
```
Skupper site controller
```
2024/08/29 11:28:11 Skupper site controller watching all namespaces
2024/08/29 11:28:11 Version: 1.5.5-rh-1
2024/08/29 11:28:11 Cluster role "skupper-service-controller-basic" created
2024/08/29 11:28:11 Starting the Skupper site controller informers
2024/08/29 11:28:11 Waiting for informer caches to sync
2024/08/29 11:28:12 Checking if sites need updates (1.5.5-rh-1)
2024/08/29 11:28:12 Starting workers
2024/08/29 11:28:12 Started workers
2024/08/29 12:01:16 Initialising skupper site ...
2024/08/29 12:01:20 Skupper site initialised
2024/08/29 12:01:20 Checking token requests...
2024/08/29 12:01:20 Done.
```

All looks good 😮

rbo, 2024-08-29:
Expected error: 
[image]

nexus-Six, 2025-01-23:
```
TASK [Fetch secret from data-center] *******************************************
FAILED - RETRYING: [localhost]: Fetch secret from data-center (10 retries left).
FAILED - RETRYING: [localhost]: Fetch secret from data-center (9 retries left).
FAILED - RETRYING: [localhost]: Fetch secret from data-center (8 retries left).
FAILED - RETRYING: [localhost]: Fetch secret from data-center (7 retries left).
FAILED - RETRYING: [localhost]: Fetch secret from data-center (6 retries left).
FAILED - RETRYING: [localhost]: Fetch secret from data-center (5 retries left).
FAILED - RETRYING: [localhost]: Fetch secret from data-center (4 retries left).
FAILED - RETRYING: [localhost]: Fetch secret from data-center (3 retries left).
FAILED - RETRYING: [localhost]: Fetch secret from data-center (2 retries left).
FAILED - RETRYING: [localhost]: Fetch secret from data-center (1 retries left).
fatal: [localhost]: FAILED! => {"api_found": true, "attempts": 10, "changed": false, "resources": [{"apiVersion": "v1", "kind": "Secret", "metadata": {"creationTimestamp": "2025-01-23T12:58:21Z", "labels": {"skupper.io/type": "connection-token-request"}, "managedFields": [{"apiVersion": "v1", "fieldsType": "FieldsV1", "fieldsV1": {"f:metadata": {"f:labels": {".": {}, "f:skupper.io/type": {}}}, "f:type": {}}, "manager": "OpenAPI-Generator", "operation": "Update", "time": "2025-01-23T12:58:21Z"}], "name": "edge-gateway", "namespace": "red-hat-service-interconnect-data-center", "resourceVersion": "585380", "uid": "ffa4918e-cf0c-46cd-8c6c-05d8ec369c75"}, "type": "Opaque"}]}

```

rbo, 2025-01-24:
Provided new `skupper-tunnel.yaml` playbook