cloud-native-robotz-hackathon/infrastructure #67: Authentication error when trying to pull the model image from DC image registry

nexus-Six, 2024-09-04:
When updating the deployment.yml for the model deployment on the robot, microshift/crio is trying to pull the image from the OCP Image Registry (e.g. default-route-openshift-image-registry.apps.cluster-xxx.com/team-1-ai/model-container:latest) but gets an unauthorized error.  Pulling the app image works.

 
[image]

rbo, 2024-09-04:
We need this : datacenter/per-user/dev/templates/rolebinding-public-pull.yaml

For ai as well.

rbo, 2024-09-04:
> We need this : datacenter/per-user/dev/templates/rolebinding-public-pull.yaml
> 
> For ai as well.
> 
> 

Here the commit https://github.com/cloud-native-robotz-hackathon/infrastructure/commit/7f3569270b3cc88f015b15df517840d36ac57284

Just copy the file into datacenter/per-user/ai/templates/ 

Argo should rollout the change direct at your running cluster.

commit 360864ef (2024-09-05): Fixes #67
closed 2024-09-05 by nexus-Six