cloud-native-robotz-hackathon/infrastructure #68: Update Microshift config basedomain to .lan

nexus-Six, 2024-09-04:
To be able to reach the Robot StarterApp from the HubCluster we need to have a domain that is resolved by the Router ( ".lan" ) and then matches the basedomain and Route of Microshift on the robot.

The default basedomain is "cluster.local" which cannot bes resolved in the router..

Apparently the basedomain can be set in the config.yaml, but there is no documentation available for 4.8 Microshift version.

rbo, 2024-09-04:
HubCluster? We have edge-gateway or data-center.

From where you want to connect to the starter app at the robot?

nexus-Six, 2024-09-04:
Sorry, from the "edge-gateway"

nexus-Six, 2024-09-04:
Current plan is to kustomize the Routes (e.g "starterapp-python-number5..." ) and then map these in the HW-Router to the robots. This way we don't need to customize all the robots.

rbo, 2024-09-04:
Adjusting the router is possible at once but not for every new workshop. 

We can create a service at the edge-gateway which point to the right robot. A service without an selector: https://kubernetes.io/docs/concepts/services-networking/service/#services-without-selectors

Means we use edge-gateway as kind of reverse proxy. Just an idea, of course not tested.

nexus-Six, 2024-09-05:
Yes, the router will be one-time setup. No changes for subsequent workshop.

Current Idea:
[Hubcontroller] (has list of robots), calls -->  http://starter-app-number5-robot-app.apps.cluster.lan/run
[Router] has a mapping of *number5* to Robot MAC adress --> forwards to Robot
[Microshift]  has and matches this route and calls the app

Issue :
Ideally we would customize the Routes in the GitOps Repo of each team to the robot name. But I believe we do not have the team to robot mapping at Env deploy time? So we will rather have to work with team routes e.g. starter-app-team1-robot-app.apps.cluster.lan/run .

rbo, 2024-09-05:
During the workshop setup (for each workshop) we update the robot <-> team mapping because of the gitops approach. 

Here is the playbook: https://github.com/cloud-native-robotz-hackathon/infrastructure/blob/main/automation/update-robot-to-team.yaml

In this playbook we can update the routes/service/reverse-proxy approach as well.

nexus-Six, 2024-09-05:
Is it feasible to push Route changes into all the team GitOps repos at that time?

rbo, 2024-09-05:
That's an option. 

Or we use an Helm chart and pass the variables via argocd application this is much easier comparing to checkout, adjust and commit N repos for each team

rbo, 2024-09-09:
I agree in 90% of cases it will be the same mapping. But I would like to be prepared if it change and it's more like IaC 😎

rbo, 2024-09-16:
I guess this is fixed with #69 ?

closed 2024-11-06 by rbo