cloud-native-robotz-hackathon/infrastructure #69: Configure Starter App Env for Robot/Microshift App Execution

nexus-Six, 2024-09-05:
While running the starter app in DevSpaces all the url Parameters are defined in config.py.

These variables need to be overwritten with the correct values in the Robot/Microshift Environment and added to the GitOps Repo.

nexus-Six, 2024-09-05:
These are primarily
ROBOT_API = ''
ROBOT_NAME = ''
INFERENCING_API = ''
INFERENCING_API_TOKEN  = ''

nexus-Six, 2024-09-12:
@rbo  We propably discussed this already, but by which URL would the microshift pod app reach the flask app (camera API) running directly on the robot host?

goetzrieger, 2024-09-12:
Currently looks like this in the image:

# cat /etc/hosts
127.0.0.1 localhost
127.0.0.1 starterapp-python-terminator.cluster.local

rbo, 2024-09-12:
> @rbo  We propably discussed this already, but by which URL would the microshift pod app reach the flask app (camera API) running directly on the robot host?

flask-app / robot-app is exposed via the route : https://github.com/cloud-native-robotz-hackathon/robot-gitops/tree/main/base/namespace/robot-app

Based on the namespace and route name it's for all robots the same. Not really useful. I agree.

Looks like we have to solve the problem next week. 

The camera api is running as systemd service? And available via hub controller (from edge-gateway) and via skupper (from data center)

rbo, 2024-09-16:
- [ ] Write `robot.local` with robot ip into `/etc/hosts` for local services (cam ip,..)
- [ ] Change MicroShift basedomain to `*.apps.$ROBOT.lan`
  - @goetzrieger please provide robot config
- [ ] Add wildcard dns entry for robots at route

rbo, 2024-09-16:
@nexus-Six `server.triton.svc.cluster.local` ports: 8000/TCP,8001/TCP,8002/TCP

rbo, 2025-02-11:
Tested. DONE

closed 2025-02-11 by rbo