cloud-native-robotz-hackathon/infrastructure #70: Fedora Detection not precise enough

nexus-Six, 2024-09-05:
For the model training, increasing the Samples (2000 is too much for the storage), batches and Epochs increases confidence score, but the coordinates always seem to be off. The square is always at the top of the hat.

  
[image]

rbo, 2024-09-05:
Is it really the object detection or the box drawing?

nexus-Six, 2024-09-05:
The mode is also trained on tennis balls. The coordinates seem to be precise.
[image]

nexus-Six, 2024-09-05:
This an image from the first trainings before the first workshop, perhaps it was always like that
[image]

nexus-Six, 2024-09-18:
Current pipeline default settings provide sufficient detection reliability. Leaving Issue open for some more testing.

closed 2024-11-06 by rbo