cloud-native-robotz-hackathon/infrastructure #86: Data Connection gets deleted when workbench gets deleted

nexus-Six, 2024-11-20:


nexus-Six, 2024-11-28:
Apparently this issue is not present in RHOAI2. 14 . Should update an test.

rbo, 2024-12-13:
Should be fixed with https://github.com/cloud-native-robotz-hackathon/infrastructure/issues/93

rbo, 2024-12-23:
Tried with OpenShift AI 2.16, created an S3 workbench and deleted it. Data connection is gone. 

#### List data connection

[image]

#### Created S3 Workbench 

[image]

#### Created a folder in S3 

[image]

#### Deleted the workbench

[image]

#### List data connection

🫨 

[image]

rbo, 2024-12-23:
Just to be clear: 

[image]

@nexus-Six 2.16 do not fix the issue. Should we create an issue at OpenShift AI folks? 

We can just provide a tooling to recover the data connection... :-/

rbo, 2024-12-23:
After a call with @mamurak, the RHODS Dashboard/WebUI deletes the related connection. I can see the delete request in the Browser web developer during the workbench deletion. And I see the request in the rhods-dashboard pod:

```
rhods-dashboard-69cd7cdd85-j6tqg rhods-dashboard {"level":30,"time":1734967310840,"pid":17,"hostname":"rhods-dashboard-69cd7cdd85-j6tqg","reqId":"req-1id","req":{"method":"DELETE","url":"/api/k8s/api/v1/namespaces/team-2-ai/secrets/data-science-pipeline-bucket-ai-connection","hostname":"rhods-dashboard-redhat-ods-applications.apps.cluster-s65ck.s65ck.sandbox2235.opentlc.com","remoteAddress":"127.0.0.1","remotePort":45826},"msg":"incoming request"}
rhods-dashboard-69cd7cdd85-j6tqg rhods-dashboard {"level":30,"time":1734967310841,"pid":17,"hostname":"rhods-dashboard-69cd7cdd85-j6tqg","msg":"Making DELETE proxy request to https://172.30.0.1:443/api/v1/namespaces/team-2-ai/secrets/data-science-pipeline-bucket-ai-connection"}
```

Max create an red hat issue for that.

Work-a-round: 
 - [ ] we add a finalizer to the secret. 
 - [ ] We create a backup secret without the data foundation annotation, in worst-case you can just copy the data.

rbo, 2024-12-23:
https://issues.redhat.com/browse/RHOAIRFE-498

rbo, 2024-12-23:
Work-a-round will be ready with:
* https://github.com/redhat-cop/agnosticd/pull/8943
* https://github.com/rhpds/agnosticv/pull/18817