cloud-native-robotz-hackathon/infrastructure #87: S3 Browser needs clarification

nexus-Six, 2024-11-20:


nexus-Six, 2024-11-28:
Visibility issue is propably due to ODF showing all buckets although you cannot see the content. Need to check ODF.

rbo, 2024-12-23:
```bash
$ oc get obc -n team-1-ai workbench-bucket -o yaml
apiVersion: objectbucket.io/v1alpha1
kind: ObjectBucketClaim
metadata:
  creationTimestamp: "2024-12-23T12:04:23Z"
  finalizers:
  - objectbucket.io/finalizer
  generation: 4
  labels:
    app: noobaa
    bucket-provisioner: openshift-storage.noobaa.io-obc
    noobaa-domain: openshift-storage.noobaa.io
  name: workbench-bucket
  namespace: team-1-ai
  resourceVersion: "88675"
  uid: b525aed2-b044-4b41-b2af-f9a4846aef95
spec:
  additionalConfig:
    bucketclass: noobaa-default-bucket-class
  bucketName: workbench-bucket-32659048-049f-4f6e-8bba-295ae5b01930
  generateBucketName: workbench-bucket
  objectBucketName: obc-team-1-ai-workbench-bucket
  storageClassName: openshift-storage.noobaa.io
status:
  phase: Bound
$ oc describe secrets -n team-1-ai workbench-bucket
Name:         workbench-bucket
Namespace:    team-1-ai
Labels:       app=noobaa
              bucket-provisioner=openshift-storage.noobaa.io-obc
              noobaa-domain=openshift-storage.noobaa.io
Annotations:  <none>

Type:  Opaque

Data
====
AWS_ACCESS_KEY_ID:      20 bytes
AWS_SECRET_ACCESS_KEY:  40 bytes

$ oc get secrets -n team-1-ai workbench-bucket \
-o go-template='[default]{{"\n"}}aws_access_key_id = {{.data.AWS_ACCESS_KEY_ID|base64decode}}{{"\n"}}aws_secret_access_key = {{.data.AWS_SECRET_ACCESS_KEY|base64decode}}{{"\n"}}' > ~/.aws/credentials

$ aws --endpoint https://s3-openshift-storage.apps.cluster-s65ck.s65ck.sandbox2235.opentlc.com s3 ls
2024-12-23 13:04:46 workbench-bucket-c7179353-110e-4901-a451-a5494599a1ed
2024-12-23 13:05:16 data-science-pipeline-buck-86572c5c-b714-4a62-a3d6-178ca77795d0
2024-12-23 13:05:09 workbench-bucket-c2210185-560b-4653-bc42-5e162607845c
2024-12-23 13:04:53 data-science-pipeline-buck-6d436936-6a51-4645-981a-49f62d6dd355
2024-12-23 13:04:30 data-science-pipeline-buck-6d821946-e062-40f0-b4a4-050169f7e487
2024-12-23 13:04:23 workbench-bucket-32659048-049f-4f6e-8bba-295ae5b01930
2024-12-23 12:28:56 first.bucket

$ aws --endpoint https://s3-openshift-storage.apps.cluster-s65ck.s65ck.sandbox2235.opentlc.com s3api get-bucket-lifecycle-configuration --bucket workbench-bucket-32659048-049f-4f6e-8bba-295ae5b01930

$ aws --endpoint https://s3-openshift-storage.apps.cluster-s65ck.s65ck.sandbox2235.opentlc.com s3api get-bucket-policy --bucket workbench-bucket-32659048-049f-4f6e-8bba-295ae5b01930

An error occurred (NoSuchBucketPolicy) when calling the GetBucketPolicy operation: The specified bucket does not have a bucket policy.

$ aws --endpoint https://s3-openshift-storage.apps.cluster-s65ck.s65ck.sandbox2235.opentlc.com s3api list-buckets
{
    "Buckets": [
        {
            "Name": "workbench-bucket-c7179353-110e-4901-a451-a5494599a1ed",
            "CreationDate": "2024-12-23T12:04:46+00:00"
        },
        {
            "Name": "data-science-pipeline-buck-86572c5c-b714-4a62-a3d6-178ca77795d0",
            "CreationDate": "2024-12-23T12:05:16+00:00"
        },
        {
            "Name": "workbench-bucket-c2210185-560b-4653-bc42-5e162607845c",
            "CreationDate": "2024-12-23T12:05:09+00:00"
        },
        {
            "Name": "data-science-pipeline-buck-6d436936-6a51-4645-981a-49f62d6dd355",
            "CreationDate": "2024-12-23T12:04:53+00:00"
        },
        {
            "Name": "data-science-pipeline-buck-6d821946-e062-40f0-b4a4-050169f7e487",
            "CreationDate": "2024-12-23T12:04:30+00:00"
        },
        {
            "Name": "workbench-bucket-32659048-049f-4f6e-8bba-295ae5b01930",
            "CreationDate": "2024-12-23T12:04:23+00:00"
        },
        {
            "Name": "first.bucket",
            "CreationDate": "2024-12-23T11:28:56+00:00"
        }
    ],
    "Owner": {
        "DisplayName": "NooBaa",
        "ID": "123"
    },
    "Prefix": null
}

$ aws --endpoint https://s3-openshift-storage.apps.cluster-s65ck.s65ck.sandbox2235.opentlc.com s3api get-bucket-acl --bucket workbench-bucket-32659048-049f-4f6e-8bba-295ae5b01930
{
    "Owner": {
        "DisplayName": "operator@noobaa.io",
        "ID": "67694979abbcbe002297a624"
    },
    "Grants": [
        {
            "Grantee": {
                "DisplayName": "operator@noobaa.io",
                "ID": "67694979abbcbe002297a624",
                "Type": "CanonicalUser"
            },
            "Permission": "FULL_CONTROL"
        }
    ]
}

```

Helpfull resources:
 * https://blog.stderr.at/openshift/2024/02/openshift-data-foundation-noobaa-bucket-data-retention-lifecycle/
 *

rbo, 2024-12-23:
I patched the S3 browser to respect the `AWS_S3_BUCKET ` https://github.com/opendatahub-io-contrib/odh-tec/issues/11

Repo: https://github.com/cloud-native-robotz-hackathon/odh-tec
ContainerImage: quay.io/cloud-native-robotz-hackathon/odh-tec:56eb8b2

closed 2025-01-10 by rbo