cloud-native-robotz-hackathon/infrastructure #111: team*-dev project shows up in the RHOAI

nexus-Six, 2025-01-10:
The -dev project now shows up in the DataScience project list in RHOAI. The visibility setting might have changed. We should hide this again, to avoid confusion.
[image]

rbo, 2025-01-10:
Looks RHOAI ignore the label "opendatahub.io/dashboard: true`

rbo, 2025-01-10:
https://redhat-internal.slack.com/archives/C07CQ4R4HMY/p1736523185140759

rbo, 2025-01-10:
@nexus-Six this is a feature, not a bug. There is no way to hide projects any more. OpenShift Project == OpenShift AI Project. 

I assign the issue to you in case you want to adjust the lab guide. If not, close it.

Thanks!

nexus-Six, 2025-01-21:
If that is the new RHOAI default, then it is fine

closed 2025-01-21 by nexus-Six