cloud-native-robotz-hackathon/infrastructure #114: Create a granular test Playbook for environment setup testing

nexus-Six, 2025-01-24:
During setup an testing it often takes lot of time to find out which component has issue. It would be great to have a Playbook that checks each component so you immediately know what to fix.

I feasible the playbook should check:

- Robot
-- Robot Control API --> e.g. http://<ROBOT_NAME>:5000/
-- Starter App API --> e.g -> http://starterapp-python.robot-app.apps.<ROBOT_NAME>
-- Triton Model Serving API --> e.g. http://triton-server.triton.apps<ROBOT_NAME>/v2/models ()

- EG
-- GitOps Operator
-- ArgoCD Applications
-- Hubcontroller API  --> e.g http://web-hub-controller.apps-crc.testing/dashboard.html
-- Hubcontroller Robot Connection --> http://web-hub-controller.apps-crc.testing/robot/status?user_key=<ROBOT_NAME> 
-- Interconnect Connection (Pods?)

- DG
-- Operators
-- Interconnect Connection (Pods?) 
-- Interconnect Route (End-to-End Test)  --> http://hub-controller-live.red-hat-service-interconnect-data-center.svc.cluster.local:8080/robot/status?user_key=<ROBOT_NAME>

rbo, 2025-01-24:
@nexus-Six EG= edge-gateway, DG=??? Data Gender? 😄

rbo, 2025-02-10:
done, let's test it tomorow @nexus-Six