cloud-native-robotz-hackathon/infrastructure #132: Make local dev setup easier - SNO or Microshift VM

goetzrieger, 2025-02-14:
Developing locally for the hackathon is cumbersome using CRC. 

Idea is to provide a pre-build KVM VM image with everything needed.

rbo, 2025-02-21:
### 4.18 not available yet:
```
[root@xps ~]# subscription-manager repos --disable=rhocp-4.18-for-rhel-9-x86_64-rpms
Repository 'rhocp-4.18-for-rhel-9-x86_64-rpms' is disabled for this system.
[root@xps ~]# subscription-manager repos --enable=rhocp-4.17-for-rhel-9-x86_64-rpms
Repository 'rhocp-4.17-for-rhel-9-x86_64-rpms' is enabled for this system.
[root@xps ~]# dnf update
Updating Subscription Management repositories.

This system has release set to 9.5 and it receives updates only for this release.

Red Hat Ansible Automation Platform 2.5 for RHEL 9 x86_64 (RPMs)                                                         3.1 MB/s | 2.2 MB     00:00
Red Hat Enterprise Linux 9 for x86_64 - BaseOS (RPMs)                                                                    8.1 MB/s |  46 MB     00:05
Fast Datapath for RHEL 9 x86_64 (RPMs)                                                                                   648 kB/s | 438 kB     00:00
Red Hat Enterprise Linux 9 for x86_64 - AppStream (RPMs)                                                                 8.2 MB/s |  50 MB     00:06
Red Hat OpenShift Container Platform 4.17 for RHEL 9 x86_64 (RPMs)                                                       6.6 MB/s |  17 MB     00:02

[root@xps ~]# dnf search microshift
Updating Subscription Management repositories.

This system has release set to 9.5 and it receives updates only for this release.

Last metadata expiration check: 0:00:38 ago on Fri 21 Feb 2025 05:36:25 AM EST.
========================================================== Name & Summary Matched: microshift ===========================================================
microshift.x86_64 : MicroShift service
microshift-greenboot.noarch : Greenboot components for MicroShift
microshift-low-latency.noarch : Baseline configuration for running low latency workload on MicroShift
microshift-multus.x86_64 : Multus CNI for MicroShift
microshift-multus-release-info.noarch : Release information for Multus CNI for MicroShift
microshift-networking.x86_64 : Networking components for MicroShift
microshift-olm.x86_64 : Operator Lifecycle Manager components for MicroShift
microshift-olm-release-info.noarch : Release information for Operator Lifecycle Manager components for MicroShift
microshift-release-info.noarch : Release information for MicroShift
microshift-selinux.noarch : SELinux policies for MicroShift

[root@xps ~]# dnf install microshift-olm.x86_64

....
```

### As root
```

dnf install microshift-olm.x86_64
cp ~robot/redhat-pullsecret-rh-ee-rbohne.json /etc/crio/openshift-pull-secret
chown root:root /etc/crio/openshift-pull-secret
chmod 600 /etc/crio/openshift-pull-secret
firewall-cmd --permanent --zone=trusted --add-source=10.42.0.0/16
firewall-cmd --permanent --zone=trusted --add-source=169.254.169.1
firewall-cmd --reload
firewall-cmd --permanent --zone=public --add-port=6443/tcp --add-port=80/tcp --add-port=443/tcp && sudo firewall-cmd --reload
systemctl enable --now microshift
```

### As robot
```
mkdir -p ~/.kube/
sudo cat /var/lib/microshift/resources/kubeadmin/kubeconfig > ~/.kube/config

oc apply -f - <<EOF
apiVersion: operators.coreos.com/v1alpha1
kind: CatalogSource
metadata:
  name: cs-redhat-operator-index
  namespace: openshift-marketplace
spec:
  image: quay.io/gpte-devops-automation/olm_snapshot_redhat_catalog:v4.16_2024_12_23
  sourceType: grpc
EOF
```

Follow https://github.com/cloud-native-robotz-hackathon/infrastructure/blob/main/docs/local-development.md#deploy--configure :
```
oc apply -k bootstrap/
# Wait for all pods are ready in openshift-gitops namespace
watch oc get pods -n openshift-gitops
```

=> Found error in `openshift-operator-lifecycle-manager/olm-operator-64b678c67d-w2tb7`

```
{"level":"error","ts":"2025-02-21T10:57:30Z","msg":"Reconciler error","controller":"operatorcondition","controllerGroup":"operators.coreos.com","controllerKind":"OperatorCondition","OperatorCondition":{"name":"openshift-gitops-operator.v1.15.0","namespace":"openshift-gitops-operator"},"namespace":"openshift-gitops-operator","name":"openshift-gitops-operator.v1.15.0","reconcileID":"09e90ff5-1c02-421c-ae59-4fe7d7bad385","error":"Deployment.apps \"openshift-gitops-operator-controller-manager\" not found","stacktrace":"sigs.k8s.io/controller-runtime/pkg/internal/controller.(*Controller).reconcileHandler\n\t/build/vendor/sigs.k8s.io/controller-runtime/pkg/internal/controller/controller.go:324\nsigs.k8s.io/controller-runtime/pkg/internal/controller.(*Controller).processNextWorkItem\n\t/build/vendor/sigs.k8s.io/controller-runtime/pkg/internal/controller/controller.go:261\nsigs.k8s.io/controller-runtime/pkg/internal/controller.(*Controller).Start.func2.2\n\t/build/vendor/sigs.k8s.io/controller-runtime/pkg/internal/controller/controller.go:222"}
```

rbo, 2025-02-21:
### Tried with an 4.17 index:
```
oc apply -f - <<EOF
apiVersion: operators.coreos.com/v1alpha1
kind: CatalogSource
metadata:
  name: cs-redhat-operator-index
  namespace: openshift-marketplace
spec:
  image: registry.redhat.io/redhat/redhat-operator-index:v4.17
  sourceType: grpc
EOF
```

ArgoCD kann not installed because:
```
      message: 'api-server resource not found installing ServiceMonitor openshift-gitops-operator-metrics-monitor:
        GroupVersionKind monitoring.coreos.com/v1, Kind=ServiceMonitor not found on
        the cluster. This API may have been deprecated and removed, see https://kubernetes.io/docs/reference/using-api/deprecation-guide/
        for more information.'
      reason: InstallComponentFailed
```

rbo, 2025-02-21:
Fetch servicemonitor CRD from other 4.17 cluster and applied to miroshift..

=> Gitops controller manager is starting...

rbo, 2025-02-21:
Bunch of errors in gitops operator:
```
025-02-21T11:17:57Z    ERROR   controller-runtime.source.EventHandler  if kind is a CRD, it should be installed before calling Start     {"kind": "Prometheus.monitoring.coreos.com", "error": "no matches for kind \"Prometheus\" in version \"monitoring.coreos.com/v1\""}
sigs.k8s.io/controller-runtime/pkg/internal/source.(*Kind).Start.func1.1
        /remote-source/deps/gomod/pkg/mod/sigs.k8s.io/controller-runtime@v0.17.2/pkg/internal/source/kind.go:63
k8s.io/apimachinery/pkg/util/wait.loopConditionUntilContext.func2
        /remote-source/deps/gomod/pkg/mod/k8s.io/apimachinery@v0.28.3/pkg/util/wait/loop.go:73
k8s.io/apimachinery/pkg/util/wait.loopConditionUntilContext
        /remote-source/deps/gomod/pkg/mod/k8s.io/apimachinery@v0.28.3/pkg/util/wait/loop.go:74
k8s.io/apimachinery/pkg/util/wait.PollUntilContextCancel
        /remote-source/deps/gomod/pkg/mod/k8s.io/apimachinery@v0.28.3/pkg/util/wait/poll.go:33
sigs.k8s.io/controller-runtime/pkg/internal/source.(*Kind).Start.func1
        /remote-source/deps/gomod/pkg/mod/sigs.k8s.io/controller-runtime@v0.17.2/pkg/internal/source/kind.go:56
```

rbo, 2025-02-21:
Because the whole clustermonitoring stack is missing

rbo, 2025-02-21:
## MicroShift is not option if we want a full argocd with an UI.

rbo, 2025-03-23:
- [ ] Deploy Hub Controller in data-center

rbo, 2025-03-24:
- [ ] Add Oauth proxy to hub-controller 
- [ ] Add NetPol to robot namespace to protect skupper services
- [ ] Update Lab guide
- [ ] Update local development guide
- [ ] Test with all 9 Robots
- [ ] Run through the lab guide