cloud-native-robotz-hackathon/infrastructure #142: Create Hyperlinks to resources explaining (Topic) in this particular case yolo5

johwes, 2025-03-04:
In the section 0. datascience there is this section,

Next, we will train our model on the basis of a Yolo5 Image Detection model to identify fedoras by providing sample images. The whole training process will run in a Pipeline leveraging OpenShift resource scaling. Sample images will be downloaded automatically and after the training, the model will be exported in onnx format to your S3 ODF (OpenShift Data Foundation) bucket. In this format we can deploy it to an inferencing service on OpenShift AI and serve it to our application.

It would be great to have a hyperlink for Yolo5 that would link to more information on what that is.

sarabanderby, 2025-03-04:
Already exists a hyperlink. Can make clearer in the lab guide as a little box, OR move it before.

rbo, 2025-03-05:
Maybe we have to add hyperlinks to every "yolo" text.

rbo, 2025-03-05:
Added some more links.

closed 2025-03-05 by rbo