cloud-native-robotz-hackathon/infrastructure #144: Make images at  Data Science part available to change and browse

rbo, 2025-03-04:
/cc @sarabanderby 

During the Data Science part it would be create to change and browse the images for training.

rbo, 2025-03-07:
@sarabanderby March 25 I meet @goetzrieger in Munich to take the pictures of the Fedora with the Robots. Can you provide us a list of images you need? Or kind of images?

- Pictures with Fedora on the ground
- Different ground flor
- Different distance
- Different light conditions
- How many?
- ...

sarabanderby, 2025-03-07:
Yes, will do! Will get back to you.

sarabanderby, 2025-03-19:
For 3 different floors, for 3 different lighting levels, for 3 different distances (make the distance within what's reasonable for the robot to do) make 4 different hat orientations.  Right side, left side, upside down and right way up. 1 picture each which makes 36 images.

goetzrieger, 2025-03-19:
So how many images in total should we take? 36 sounds a bit too few...

@sarabanderby what tool should we use for annotating? I found an app called Labelimg which can do YOLO style labelling. But it segfaulted on my first try drawing a box. Sigh.

Do you have any other recommendations?

sarabanderby, 2025-03-20:
Label studio works and can de deployed in open shift :D 
You can do more angles, so do 5 or 6. But in the demo we download a 100 images for training, using 50 specific instead of 100 random MIGHT be better but needs to be tested.

rbo, 2025-03-20:
Should we consider to deploy Label studio by default for an optional labeling part?

sarabanderby, 2025-03-20:
Intersting, having an automated labeling pipeline would give us the option to let participants generate their own data sets in the hackathon as well.

goetzrieger, 2025-03-20:
> Label studio works and can de deployed in open shift :D You can do more angles, so do 5 or 6. But in the demo we download a 100 images for training, using 50 specific instead of 100 random MIGHT be better but needs to be tested.

So we should still aim for 100 I'd say.

rbo, 2025-07-15:
might be ducplicated to #185

closed 2025-10-23 by goetzrieger