cloud-native-robotz-hackathon/infrastructure #154: Include a LLM code assistent in dev spaces

rbo, 2025-03-07:
Feedback from Stockholm: 
https://github.com/cloud-native-robotz-hackathon/infrastructure/issues/152

Let's dicuss, Manual had a dev spaces "code support" solution / demo.

nexus-Six, 2025-08-20:
I was able to deploy Codellama 7B to RHOAI as vLLm inferencing server. I can connect my local VSCode / Continue without problem. But when I install the continue extension from the VSCode extension catalog into DevSpaces and setup the config.yaml it always crashes. 

I though I saw somewhere the default Continue extension is not compatible with UDI/RHEL and you have to build it yourself? @manhah Do you have any idea on this?

nexus-Six, 2025-08-20:
@rbo We need to prepare team ConfigMaps that contain a config.json, containing the Codellama inferencing URL ( e.g. https://codellama2-codeassistant.apps.cluster-xxx.opentlc.com/v1 ), which are then injexted via devfile into ~/.continue/config.json. I assume this should not be a problem?

nexus-Six, 2025-08-20:
Python Script for Model Download (set HF Token) 

[download_codellama.py](https://github.com/user-attachments/files/21894761/download_codellama.py)

nexus-Six, 2025-09-08:
The script is inked up in the comments : https://github.com/cloud-native-robotz-hackathon/infrastructure/issues/154#issuecomment-3206063540

rbo, 2025-09-09:
Following steps to solve get the model stuff running:
* enable kserve
* Installed Serverless Operator
* Installed Service Mesh Operator
* Installed Authorino Operator - let's try it again without.

[image]

nexus-Six, 2025-09-17:
Fixed file permission issue with continue config.json by copying it.  Looks good from my side in DevSpaces.

nexus-Six, 2025-09-17:
[image]