cloud-native-robotz-hackathon/infrastructure #161: WebSocket should use ssl and not non-ssl connection

rbo, 2025-03-25:
[image]

```
./src/test/mockserver.properties:mockserver.maxSocketTimeout=120000
./src/main/resources/META-INF/resources/js/dashboard.js:var socket;
./src/main/resources/META-INF/resources/js/dashboard.js:    socket = new WebSocket("ws://" + location.host + "/dashboard/" + clientId);
./src/main/resources/META-INF/resources/js/dashboard.js:    socket.onopen = function () {
./src/main/resources/META-INF/resources/js/dashboard.js:
```

=> `wss://` intead of `ws://`?

Part of: #132

nexus-Six, 2025-04-17:
Fixed in https://github.com/cloud-native-robotz-hackathon/hubcontroller-rest-quarkus/commit/dd37e8dc3632190a9e9a72092c90fb5da5f6e451

closed 2025-04-17 by nexus-Six

nexus-Six, 2025-07-15:
Testing