cloud-native-robotz-hackathon/infrastructure #184: Upgrade data center env. to OpenShift v4.19

rbo, 2025-07-01:


rbo, 2025-07-01:
Development tile failed to deploy:

```
{
  "exception": "Traceback (most recent call last):\n  File \"/tmp/ansible_kubernetes.core.k8s_info_payload_bguga6tj/ansible_kubernetes.core.k8s_info_payload.zip/ansible_collections/kubernetes/core/plugins/modules/k8s_info.py\", line 224, in main\n  File \"/tmp/ansible_kubernetes.core.k8s_info_payload_bguga6tj/ansible_kubernetes.core.k8s_info_payload.zip/ansible_collections/kubernetes/core/plugins/modules/k8s_info.py\", line 184, in execute_module\n  File \"/tmp/ansible_kubernetes.core.k8s_info_payload_bguga6tj/ansible_kubernetes.core.k8s_info_payload.zip/ansible_collections/kubernetes/core/plugins/module_utils/k8s/service.py\", line 322, in find\n    raise CoreException(\nansible_collections.kubernetes.core.plugins.module_utils.k8s.exceptions.CoreException: Failed to gather information about StorageSystem(s) even after waiting for 600 seconds\n",
  "msg": "Failed to gather information about StorageSystem(s) even after waiting for 600 seconds",
  "invocation": {
    "module_args": {
      "api_version": "odf.openshift.io/v1alpha1",
      "kind": "StorageSystem",
      "name": "ocs-storagecluster-storagesystem",
      "namespace": "openshift-storage",
      "wait": true,
      "wait_condition": {
        "reason": "ReconcileCompleted",
        "status": true,
        "type": "Available"
      },
      "wait_sleep": 10,
      "wait_timeout": 600,
      "label_selectors": [],
      "field_selectors": [],
      "kubeconfig": null,
      "context": null,
      "host": null,
      "api_key": null,
      "username": null,
      "password": null,
      "validate_certs": null,
      "ca_cert": null,
      "client_cert": null,
      "client_key": null,
      "proxy": null,
      "no_proxy": null,
      "proxy_headers": null,
      "persist_config": null,
      "impersonate_user": null,
      "impersonate_groups": null,
      "hidden_fields": null
    }
  },
  "_ansible_no_log": false,
  "changed": false
}
```

sarabanderby, 2025-07-02:
The computer says no

[image]

rbo, 2025-07-03:
Mh, for example team-1 looks like have the right permissions:

[image]

rbo, 2025-07-03:
Created a branch `issue-184` in
 * https://github.com/cloud-native-robotz-hackathon/infrastructure
 * https://github.com/cloud-native-robotz-hackathon/lab-guide
for the update and deployment.

sarabanderby, 2025-07-04:
@rbo 
I get this funny "deleted" in the UI. The notebook runs fine, its seems to a warning. It gives the option to "update" but it would be great if this wasn't here so we don't confuse the users. First glance it looks like deployment failed. 

[image]

rbo, 2025-07-07:
@sarabanderby fixed:

[image]

nexus-Six, 2025-07-15:
Labguide needs to be fixed (Branch issue-184)

rbo, 2025-07-16:
> Now getting this in the NVidia operator pods and crashing :
> 
> Error: Unable to find a match: elfutils-libelf-devel.x86_64 FATAL: failed to install elfutils packages. RHEL entitlement may be improperly deployed.
> 
> * echo 'FATAL: failed to install elfutils packages. RHEL entitlement may be improperly deployed.'

🤔 might be an issue related to the rhel change. 4.18 -> RHEL 9.4, 4.19 -> RHEL 9.6

I have to check

rbo, 2025-07-21:
Some pods are missing ;-) 
```
bash-5.1 ~ $ oc get pods -n nvidia-gpu-operator 
NAME                            READY   STATUS    RESTARTS   AGE
gpu-operator-6974b5d588-6clxx   1/1     Running   0          78m
bash-5.1 ~ $ 
```

rbo, 2025-07-21:
Looks like NFD is missing, nodes have less labels as expected:

```
bash-5.1 ~ $ oc get nodes --show-labels
NAME                                        STATUS   ROLES                  AGE    VERSION   LABELS
ip-10-0-16-82.eu-west-1.compute.internal    Ready    worker,worker-gpu      83m    v1.32.5   beta.kubernetes.io/arch=amd64,beta.kubernetes.io/instance-type=g5.xlarge,beta.kubernetes.io/os=linux,failure-domain.beta.kubernetes.io/region=eu-west-1,failure-domain.beta.kubernetes.io/zone=eu-west-1a,kubernetes.io/arch=amd64,kubernetes.io/hostname=ip-10-0-16-82.eu-west-1.compute.internal,kubernetes.io/os=linux,node-role.kubernetes.io/worker-gpu=,node-role.kubernetes.io/worker=,node.kubernetes.io/instance-type=g5.xlarge,node.openshift.io/os_id=rhel,topology.ebs.csi.aws.com/zone=eu-west-1a,topology.k8s.aws/zone-id=euw1-az3,topology.kubernetes.io/region=eu-west-1,topology.kubernetes.io/zone=eu-west-1a
ip-10-0-20-158.eu-west-1.compute.internal   Ready    arm,worker             81m    v1.32.5   beta.kubernetes.io/arch=arm64,beta.kubernetes.io/instance-type=m7g.2xlarge,beta.kubernetes.io/os=linux,failure-domain.beta.kubernetes.io/region=eu-west-1,failure-domain.beta.kubernetes.io/zone=eu-west-1a,kubernetes.io/arch=arm64,kubernetes.io/hostname=ip-10-0-20-158.eu-west-1.compute.internal,kubernetes.io/os=linux,node-role.kubernetes.io/arm=,node-role.kubernetes.io/worker=,node.kubernetes.io/instance-type=m7g.2xlarge,node.openshift.io/os_id=rhel,topology.ebs.csi.aws.com/zone=eu-west-1a,topology.k8s.aws/zone-id=euw1-az3,topology.kubernetes.io/region=eu-west-1,topology.kubernetes.io/zone=eu-west-1a
ip-10-0-3-215.eu-west-1.compute.internal    Ready    worker                 103m   v1.32.5   beta.kubernetes.io/arch=amd64,beta.kubernetes.io/instance-type=m5.8xlarge,beta.kubernetes.io/os=linux,cluster.ocs.openshift.io/openshift-storage=,failure-domain.beta.kubernetes.io/region=eu-west-1,failure-domain.beta.kubernetes.io/zone=eu-west-1a,kubernetes.io/arch=amd64,kubernetes.io/hostname=ip-10-0-3-215.eu-west-1.compute.internal,kubernetes.io/os=linux,node-role.kubernetes.io/worker=,node.kubernetes.io/instance-type=m5.8xlarge,node.openshift.io/os_id=rhel,topology.ebs.csi.aws.com/zone=eu-west-1a,topology.k8s.aws/zone-id=euw1-az3,topology.kubernetes.io/region=eu-west-1,topology.kubernetes.io/zone=eu-west-1a,topology.rook.io/rack=rack2
ip-10-0-31-0.eu-west-1.compute.internal     Ready    worker                 103m   v1.32.5   beta.kubernetes.io/arch=amd64,beta.kubernetes.io/instance-type=m5.8xlarge,beta.kubernetes.io/os=linux,cluster.ocs.openshift.io/openshift-storage=,failure-domain.beta.kubernetes.io/region=eu-west-1,failure-domain.beta.kubernetes.io/zone=eu-west-1a,kubernetes.io/arch=amd64,kubernetes.io/hostname=ip-10-0-31-0.eu-west-1.compute.internal,kubernetes.io/os=linux,node-role.kubernetes.io/worker=,node.kubernetes.io/instance-type=m5.8xlarge,node.openshift.io/os_id=rhel,topology.ebs.csi.aws.com/zone=eu-west-1a,topology.k8s.aws/zone-id=euw1-az3,topology.kubernetes.io/region=eu-west-1,topology.kubernetes.io/zone=eu-west-1a,topology.rook.io/rack=rack2
ip-10-0-37-36.eu-west-1.compute.internal    Ready    control-plane,master   117m   v1.32.5   beta.kubernetes.io/arch=amd64,beta.kubernetes.io/instance-type=m5d.2xlarge,beta.kubernetes.io/os=linux,failure-domain.beta.kubernetes.io/region=eu-west-1,failure-domain.beta.kubernetes.io/zone=eu-west-1a,kubernetes.io/arch=amd64,kubernetes.io/hostname=ip-10-0-37-36.eu-west-1.compute.internal,kubernetes.io/os=linux,node-role.kubernetes.io/control-plane=,node-role.kubernetes.io/master=,node.kubernetes.io/instance-type=m5d.2xlarge,node.openshift.io/os_id=rhel,topology.ebs.csi.aws.com/zone=eu-west-1a,topology.k8s.aws/zone-id=euw1-az3,topology.kubernetes.io/region=eu-west-1,topology.kubernetes.io/zone=eu-west-1a
ip-10-0-41-12.eu-west-1.compute.internal    Ready    worker                 103m   v1.32.5   beta.kubernetes.io/arch=amd64,beta.kubernetes.io/instance-type=m5.8xlarge,beta.kubernetes.io/os=linux,cluster.ocs.openshift.io/openshift-storage=,failure-domain.beta.kubernetes.io/region=eu-west-1,failure-domain.beta.kubernetes.io/zone=eu-west-1a,kubernetes.io/arch=amd64,kubernetes.io/hostname=ip-10-0-41-12.eu-west-1.compute.internal,kubernetes.io/os=linux,node-role.kubernetes.io/worker=,node.kubernetes.io/instance-type=m5.8xlarge,node.openshift.io/os_id=rhel,topology.ebs.csi.aws.com/zone=eu-west-1a,topology.k8s.aws/zone-id=euw1-az3,topology.kubernetes.io/region=eu-west-1,topology.kubernetes.io/zone=eu-west-1a,topology.rook.io/rack=rack0
ip-10-0-46-36.eu-west-1.compute.internal    Ready    worker                 103m   v1.32.5   beta.kubernetes.io/arch=amd64,beta.kubernetes.io/instance-type=m5.8xlarge,beta.kubernetes.io/os=linux,cluster.ocs.openshift.io/openshift-storage=,failure-domain.beta.kubernetes.io/region=eu-west-1,failure-domain.beta.kubernetes.io/zone=eu-west-1a,kubernetes.io/arch=amd64,kubernetes.io/hostname=ip-10-0-46-36.eu-west-1.compute.internal,kubernetes.io/os=linux,node-role.kubernetes.io/worker=,node.kubernetes.io/instance-type=m5.8xlarge,node.openshift.io/os_id=rhel,topology.ebs.csi.aws.com/zone=eu-west-1a,topology.k8s.aws/zone-id=euw1-az3,topology.kubernetes.io/region=eu-west-1,topology.kubernetes.io/zone=eu-west-1a,topology.rook.io/rack=rack0
ip-10-0-48-34.eu-west-1.compute.internal    Ready    arm,worker             81m    v1.32.5   beta.kubernetes.io/arch=arm64,beta.kubernetes.io/instance-type=m7g.2xlarge,beta.kubernetes.io/os=linux,failure-domain.beta.kubernetes.io [... cut ...] .io/region=eu-west-1,failure-domain.beta.kubernetes.io/zone=eu-west-1a,kubernetes.io/arch=amd64,kubernetes.io/hostname=ip-10-0-63-198.eu-west-1.compute.internal,kubernetes.io/os=linux,node-role.kubernetes.io/control-plane=,node-role.kubernetes.io/master=,node.kubernetes.io/instance-type=m5d.2xlarge,node.openshift.io/os_id=rhel,topology.ebs.csi.aws.com/zone=eu-west-1a,topology.k8s.aws/zone-id=euw1-az3,topology.kubernetes.io/region=eu-west-1,topology.kubernetes.io/zone=eu-west-1a
bash-5.1 ~ $ 
```

rbo, 2025-07-21:
```
bash-5.1 ~ $ oc get pods -n openshift-nfd
NAME                                      READY   STATUS    RESTARTS       AGE
nfd-controller-manager-7f759b464f-w6snj   1/1     Running   0              82m
nfd-gc-d7878f4bb-gzbsb                    1/1     Running   0              81m
nfd-master-9b6f7bf4c-2vpv2                0/1     Running   16 (94s ago)   81m
nfd-worker-2qdgx                          1/1     Running   0              81m
nfd-worker-5t8xj                          1/1     Running   0              81m
nfd-worker-d846s                          1/1     Running   0              81m
nfd-worker-f6fj8                          1/1     Running   0              81m
nfd-worker-hxxfm                          1/1     Running   0              81m
nfd-worker-kpjxs                          1/1     Running   0              81m
nfd-worker-mfmsz                          1/1     Running   0              81m
nfd-worker-svnlc                          1/1     Running   0              81m
nfd-worker-vlf8l                          1/1     Running   0              81m
bash-5.1 ~ $ oc get NodeFeatureDiscovery -A -o yaml
apiVersion: v1
items:
- apiVersion: nfd.openshift.io/v1
  kind: NodeFeatureDiscovery
  metadata:
    creationTimestamp: "2025-07-21T15:00:22Z"
    finalizers:
    - nfd-finalizer
    generation: 2
    name: nfd-instance
    namespace: openshift-nfd
    resourceVersion: "49068"
    uid: 84ce74ab-9467-409c-b7e2-ca0948616ffe
  spec:
    enableTaints: false
    instance: ""
    operand:
      image: registry.redhat.io/openshift4/ose-node-feature-discovery-rhel9:v4.17
      imagePullPolicy: IfNotPresent
      servicePort: 12000
    prunerOnDelete: false
    topologyUpdater: false
    workerConfig:
      configData: |
        core:
        #  labelWhiteList:
        #  noPublish: false
          sleepInterval: 60s
        #  sources: [all]
        #  klog:
        #    addDirHeader: false
        #    alsologtostderr: false
        #    logBacktraceAt:
        #    logtostderr: true
        #    skipHeaders: false
        #    stderrthreshold: 2
        #    v: 0
        #    vmodule:
        ##   NOTE: the following options are not dynamically run-time
        ##          configurable and require a nfd-worker restart to take effect
        ##          after being changed
        #    logDir:
        #    logFile:
        #    logFileMaxSize: 1800
        #    skipLogHeaders: false
        sources:
        #  cpu:
        #    cpuid:
        ##     NOTE: whitelist has priority over blacklist
        #      attributeBlacklist:
        #        - "BMI1"
        #        - "BMI2"
        #        - "CLMUL"
        #        - "CMOV"
        #        - "CX16"
        #        - "ERMS"
        #        - "F16C"
        #        - "HTT"
        #        - "LZCNT"
        #        - "MMX"
        #        - "MMXEXT"
        #        - "NX"
        #        - "POPCNT"
        #        - "RDRAND"
        #        - "RDSEED"
        #        - "RDTSCP"
        #        - "SGX"
        #        - "SSE"
        #        - "SSE2"
        #        - "SSE3"
        #        - "SSE4.1"
        #        - "SSE4.2"
        #        - "SSSE3"
        #      attributeWhitelist:
        #  kernel:
        #    kconfigFile: "/path/to/kconfig"
        #    configOpts:
        #      - "NO_HZ"
        #      - "X86"
        #      - "DMI"
          pci:
            deviceClassWhitelist:
              - "0200"
              - "03"
              - "12"
            deviceLabelFields:
        #      - "class"
              - "vendor"
        #      - "device"
        #      - "subsystem_vendor"
        #      - "subsystem_device"
        #  usb:
        #    deviceClassWhitelist:
        #      - "0e"
        #      - "ef"
        #      - "fe"
        #      - "ff"
        #    deviceLabelFields:
        #      - "class"
        #      - "vendor"
        #      - "device"
        #  custom:
        #    - name: "my.kernel.feature"
        #      matchOn:
        #        - loadedKMod: ["example_kmod1", "example_kmod2"]
        #    - name: "my.pci.feature"
        #      matchOn:
        #        - pciId:
        #            class: ["0200"]
        #            vendor: ["15b3"]
        #            device: ["1014", "1017"]
        #        - pciId :
        #            vendor: ["8086"]
        #            device: ["1000", "1100"]
        #    - name: "my.usb.feature"
        #      matchOn:
        #        - usbId:
        #          class: ["ff"]
        #          vendor: ["03e7"]
        #          device: ["2485"]
        #        - usbId:
        #          class: ["fe"]
        #          vendor: ["1a6e"]
        #          device: ["089a"]
        #    - name: "my.combined.feature"
        #      matchOn:
        #        - pciId:
        #            vendor: ["15b3"]
        #            device: ["1014", "1017"]
        #          loadedKMod : ["vendor_kmod1", "vendor_kmod2"]
  status:
    conditions:
    - lastTransitionTime: "2025-07-21T15:00:32Z"
      message: ""
      reason: ConditionNotBeingMetCurrently
      status: "False"
      type: Available
    - lastTransitionTime: "2025-07-21T15:00:32Z"
      message: ""
      reason: ConditionNotBeingMetCurrently
      status: "False"
      type: Upgradeable
    - lastTransitionTime: "2025-07-21T15:00:32Z"
      message: ""
      reason: ConditionNotBeingMetCurrently
      status: "False"
      type: Progressing
    - lastTransitionTime: "2025-07-21T15:00:32Z"
      message: number of available pods is 0
      reason: NFDMasterDeploymentDegraded
      status: "True"
      type: Degraded
kind: List
metadata:
  resourceVersion: ""
bash-5.1 ~ $ 
```

rbo, 2025-07-21:
Created a new NodeFeatureDiscovery CR
```yaml
apiVersion: nfd.openshift.io/v1
kind: NodeFeatureDiscovery
metadata:
  creationTimestamp: '2025-07-21T16:24:01Z'
  finalizers:
    - nfd-finalizer
  generation: 2
  managedFields:
    - apiVersion: nfd.openshift.io/v1
      fieldsType: FieldsV1
      fieldsV1:
        'f:spec':
          .: {}
          'f:operand':
            .: {}
            'f:imagePullPolicy': {}
            'f:servicePort': {}
          'f:workerConfig':
            .: {}
            'f:configData': {}
      manager: Mozilla
      operation: Update
      time: '2025-07-21T16:24:01Z'
    - apiVersion: nfd.openshift.io/v1
      fieldsType: FieldsV1
      fieldsV1:
        'f:metadata':
          'f:finalizers':
            .: {}
            'v:"nfd-finalizer"': {}
        'f:spec':
          'f:enableTaints': {}
          'f:instance': {}
          'f:prunerOnDelete': {}
          'f:topologyUpdater': {}
      manager: node-feature-discovery-operator
      operation: Update
      time: '2025-07-21T16:24:01Z'
    - apiVersion: nfd.openshift.io/v1
      fieldsType: FieldsV1
      fieldsV1:
        'f:status':
          .: {}
          'f:conditions': {}
      manager: node-feature-discovery-operator
      operation: Update
      subresource: status
      time: '2025-07-21T16:24:01Z'
  name: nfd-instance
  namespace: openshift-nfd
  resourceVersion: '167920'
  uid: 41f613f4-4a3d-4dfb-a77c-7728894d511c
spec:
  enableTaints: false
  instance: ''
  operand:
    imagePullPolicy: IfNotPresent
    servicePort: 12000
  prunerOnDelete: false
  topologyUpdater: false
  workerConfig:
    configData: |
      core:
      #  labelWhiteList:
      #  noPublish: false
        sleepInterval: 60s
      #  sources: [all]
      #  klog:
      #    addDirHeader: false
      #    alsologtostderr: false
      #    logBacktraceAt:
      #    logtostderr: true
      #    skipHeaders: false
      #    stderrthreshold: 2
      #    v: 0
      #    vmodule:
      ##   NOTE: the following options are not dynamically run-time 
      ##          configurable and require a nfd-worker restart to take effect
      ##          after being changed
      #    logDir:
      #    logFile:
      #    logFileMaxSize: 1800
      #    skipLogHeaders: false
      sources:
      #  cpu:
      #    cpuid:
      ##     NOTE: whitelist has priority over blacklist
      #      attributeBlacklist:
      #        - "BMI1"
      #        - "BMI2"
      #        - "CLMUL"
      #        - "CMOV"
      #        - "CX16"
      #        - "ERMS"
      #        - "F16C"
      #        - "HTT"
      #        - "LZCNT"
      #        - "MMX"
      #        - "MMXEXT"
      #        - "NX"
      #        - "POPCNT"
      #        - "RDRAND"
      #        - "RDSEED"
      #        - "RDTSCP"
      #        - "SGX"
      #        - "SSE"
      #        - "SSE2"
      #        - "SSE3"
      #        - "SSE4.1"
      #        - "SSE4.2"
      #        - "SSSE3"
      #      attributeWhitelist:
      #  kernel:
      #    kconfigFile: "/path/to/kconfig"
      #    configOpts:
      #      - "NO_HZ"
      #      - "X86"
      #      - "DMI"
        pci:
          deviceClassWhitelist:
            - "0200"
            - "03"
            - "12"
          deviceLabelFields:
      #      - "class"
            - "vendor"
      #      - "device"
      #      - "subsystem_vendor"
      #      - "subsystem_device"
      #  usb:
      #    deviceClassWhitelist:
      #      - "0e"
      #      - "ef"
      #      - "fe"
      #      - "ff"
      #    deviceLabelFields:
      #      - "class"
      #      - "vendor"
      #      - "device"
      #  custom:
      #    - name: "my.kernel.feature"
      #      matchOn:
      #        - loadedKMod: ["example_kmod1", "example_kmod2"]
      #    - name: "my.pci.feature"
      #      matchOn:
      #        - pciId:
      #            class: ["0200"]
      #            vendor: ["15b3"]
      #            device: ["1014", "1017"]
      #        - pciId :
      #            vendor: ["8086"]
      #            device: ["1000", "1100"]
      #    - name: "my.usb.feature"
      #      matchOn:
      #        - usbId:
      #          class: ["ff"]
      #          vendor: ["03e7"]
      #          device: ["2485"]
      #        - usbId:
      #          class: ["fe"]
      #          vendor: ["1a6e"]
      #          device: ["089a"]
      #    - name: "my.combined.feature"
      #      matchOn:
      #        - pciId:
      #            vendor: ["15b3"]
      #            device: ["1014", "1017"]
      #          loadedKMod : ["vendor_kmod1", "vendor_kmod2"]
status:
  conditions:
    - lastTransitionTime: '2025-07-21T16:24:01Z'
      message: ''
      reason: ConditionNotBeingMetCurrently
      status: 'False'
      type: Available
    - lastTransitionTime: '2025-07-21T16:24:01Z'
      message: ''
      reason: ConditionNotBeingMetCurrently
      status: 'False'
      type: Upgradeable
    - lastTransitionTime: '2025-07-21T16:24:01Z'
      message: ds is progressing
      reason: NFDWorkerDaemonSetProgressing
      status: 'True'
      type: Progressing
    - lastTransitionTime: '2025-07-21T16:24:01Z'
      message: ''
      reason: ConditionNotBeingMetCurrently
      status: 'False'
      type: Degraded

```

rbo, 2025-07-25:
Successful deployment of prod tile.

DONE

closed 2025-07-25 by rbo