cloud-native-robotz-hackathon/infrastructure #185: People would like to see downloaded images, understand and monitor what is happening

nexus-Six, 2025-07-15:


rbo, 2025-07-22:
I created a PVC:

```
kind: PersistentVolumeClaim
apiVersion: v1
metadata:
  name: object-detection-training-pvc
  labels:
    opendatahub.io/dashboard: 'true'
spec:
  accessModes:
    - ReadWriteOnce
  resources:
    requests:
      storage: 10Gi
  storageClassName: gp3-csi
  volumeMode: Filesystem
```

Attached the PVC to Workbench:

[image]

It's not visible in the juptier notebook. 

⚠️ **Problem**: gp3-csi support only RWO means workbench and pipeline run's on the same node. That could case strange problems.

rbo, 2025-07-22:
Running GPU Pipeline cephfs make storageproblems:
```
MountVolume.MountDevice failed for volume "pvc-b02269b1-9f89-4ddc-8904-efd09ff0f649" : kubernetes.io/csi: attacher.MountDevice failed to create newCsiDriverClient: driver name openshift-storage.cephfs.csi.ceph.com not found in the list of registered CSI drivers
```

rbo, 2025-07-22:
> ⚠️ Limits in the pipeline are to high: 24G & 8 Cores Temp. Changed to 12 GB & * Cores because we only have a g5.xlarge instance

Fixed in repo. Should be fine with next env.